zephyrproject-rtos/zephyr #335: images for the wiki

nashif, 2017-05-30:
Use this issue to add images that can be linked from the wiki. There is no way to upload images to the wiki right now.

nashif, 2017-05-30:
[image]

dbkinder, 2017-05-31:
This option wasn't availble because the issue tab wasn't enabled previously. Indeed uploadng an image to the issue (drag and drop) is one oe the "cheats" for getting an url for uploading images to the wiki, but the image upload file won't be part of the wiki git repo (it's part of the issue).

nashif, 2017-05-31:
and that is fully ok, the link will still be available in the git history, having the file in the wiki git is not critical. We will still have the history of the image changes in git.

mjlinkme, 2017-06-09:
[image]

mjlinkme, 2017-06-09:
[image]

dbkinder, 2017-08-17:
[image]
[image]
[image]
[image]
[image]

linkmeyer, 2017-12-14:
[More Schedule Details.pdf](https://github.com/zephyrproject-rtos/zephyr/files/1561049/More.Schedule.Details.pdf)

dbkinder, 2018-01-23:
Issue needs to remain open for us to upload images for the wiki

dbkinder, 2018-01-23:
[image]
[image]

MaureenHelm, 2018-05-04:
Is this issue still needed? Why not commit the images to the wiki git?

dbkinder, 2018-07-17:
We created this so folks without merge permission could upload images here, and then reference them in the wiki.  We can close this.

nashif, 2018-08-15:
[image]
[image]

nashif, 2019-04-11:
[image]

nashif, 2019-04-11:
[image]

dbkinder, 2019-05-24:
[image]
[image]

dbkinder, 2019-05-24:
added some images to the wiki, so can close this issue again.

closed 2019-05-24 by dbkinder