zephyrproject-rtos/zephyr #372: Error Flasing Nucleo F401RE

lucastanure, 2017-06-01:
System : Arch Linux

Procedure : 
Install sdk https://www.zephyrproject.org/doc/getting_started/installation_linux.html
$ git clone https://github.com/zephyrproject-rtos/zephyr.git
$ cd <zephyr_root_path>
$ source zephyr-env.sh
$ cd $ZEPHYR_BASE/samples/hello_world/
$ make BOARD=nucleo_f401re
$ make BOARD=nucleo_f401re flash

Output:
tanure@archTanure  hello_world $ make BOARD=nucleo_f401re flash
make[1]: Entering directory '/home/tanure/workspace/zephyr'
make[2]: Entering directory '/home/tanure/workspace/zephyr/samples/hello_world/outdir/nucleo_f401re'
  Using /home/tanure/workspace/zephyr as source for kernel
  GEN     ./Makefile
  CHK     include/generated/version.h
  CHK     include/generated/generated_dts_board.h
  CHK     misc/generated/configs.c
  CHK     include/generated/offsets.h
make[3]: 'isr_tables.o' is up to date.
Flashing nucleo_f401re
Flashing Target Device
Open On-Chip Debugger 0.9.0-dirty (2017-05-16-18:46)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v27 API v2 SWIM v15 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.257255
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f4x.cpu       hla_target little stm32f4x.cpu       halted
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000358 msp: 0x20018000
auto erase enabled
Info : device id = 0x10016433
Info : flash size = 512kbytes
**Warn : no flash bank found for address 0**
wrote 0 bytes from file /home/tanure/workspace/zephyr/samples/hello_world/outdir/nucleo_f401re/zephyr.bin in 0.000737s (0.000 KiB/s)
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000358 msp: 0x20018000
diff 0 address 0x00000000. Was 0x00 instead of 0x68
diff 1 address 0x00000001. Was 0x80 instead of 0x08
diff 2 address 0x00000002. Was 0x01 instead of 0x00
diff 3 address 0x00000004. Was 0x59 instead of 0x1d
diff 4 address 0x00000005. Was 0x03 instead of 0x21
diff 5 address 0x00000008. Was 0xa1 instead of 0x6d
diff 6 address 0x00000009. Was 0x03 instead of 0x21
diff 7 address 0x0000000c. Was 0xa1 instead of 0xc5
More than 8 errors, the rest are not printed.

**make[2]: *** [/home/tanure/workspace/zephyr/Makefile:1299: flash] Error 1
make[2]: Leaving directory '/home/tanure/workspace/zephyr/samples/hello_world/outdir/nucleo_f401re'
make[1]: *** [Makefile:176: sub-make] Error 2
make[1]: Leaving directory '/home/tanure/workspace/zephyr'
make: *** [/home/tanure/workspace/zephyr/Makefile.inc:88: flash] Error 2**

nashif, 2017-06-01:
The openocd version on the SDK is most likely not compatible and does not have the needed support for this board, did you try with latest openocd (0.10)?

lucastanure, 2017-06-01:
With openocd 0.10 I have one error :

[tanure@archTanure tanure]# openocd -f /usr/share/openocd/scripts/board/stm32f4discovery.cfg 
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
**Error: open failed**
in procedure 'init' 
in procedure 'ocd_bouncer'

nashif, 2017-06-01:
did you try 'make ... flash'?

lucastanure, 2017-06-01:
Wrong config file .I should use this with the new openocd:
openocd -f /usr/share/openocd/scripts/board/st_nucleo_f4.cfg

nashif, 2017-06-01:
right, this is the one used here: boards/arm/nucleo_f401re/support/openocd.cfg

lucastanure, 2017-06-01:
tanure@archTanure  hello_world $ export OPENOCD_DEFAULT_PATH=/usr/share/openocd 
tanure@archTanure  hello_world $ make BOARD=nucleo_f401re flash 
make[1]: Entering directory '/home/tanure/workspace/zephyr'
make[2]: Entering directory '/home/tanure/workspace/zephyr/samples/hello_world/outdir/nucleo_f401re'
  Using /home/tanure/workspace/zephyr as source for kernel
  GEN     ./Makefile
  CHK     include/generated/version.h
  CHK     include/generated/generated_dts_board.h
  CHK     misc/generated/configs.c
  CHK     include/generated/offsets.h
make[3]: 'isr_tables.o' is up to date.
Flashing nucleo_f401re
Flashing Target Device
/home/tanure/workspace/zephyr/boards/arm/nucleo_f401re/support/openocd.cfg
Open On-Chip Debugger 0.9.0-dirty (2017-05-16-18:46)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
/home/tanure/workspace/zephyr/boards/arm/nucleo_f401re/support/openocd.cfg:1: Error: Can't find board/st_nucleo_f4.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
at file "/home/tanure/workspace/zephyr/boards/arm/nucleo_f401re/support/openocd.cfg", line 1
make[2]: *** [/home/tanure/workspace/zephyr/Makefile:1299: flash] Error 1
make[2]: Leaving directory '/home/tanure/workspace/zephyr/samples/hello_world/outdir/nucleo_f401re'
make[1]: *** [Makefile:176: sub-make] Error 2
make[1]: Leaving directory '/home/tanure/workspace/zephyr'
make: *** [/home/tanure/workspace/zephyr/Makefile.inc:88: flash] Error 2

lucastanure, 2017-06-01:
Openocd 0.10 with this board works.

lucastanure, 2017-06-01:
Fix : 
Install the required packages in a Arch Linux host system with:
confuse  hidapi  libftdi  libftdi-compat  libusb-compat

closed 2017-06-01 by lucastanure