zephyrproject-rtos/zephyr #560: arduino_101_ble fails to build samples/bluetooth/hci_uart/test_nrf5

galak, 2017-06-21:
When running sanity check we get:

arduino_101_ble           samples/bluetooth/hci_uart/test_nrf5               FAILED: build_error
	see: sanity-out/arduino_101_ble/samples/bluetooth/hci_uart/test_nrf5/build.log

seems the test is too big now for arduino_101_ble:

  LINK    zephyr.lnk
/opt/zephyr-sdk-0.9.1/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: zephyr_prebuilt.elf section `noinit' will not fit in region `SRAM'
/opt/zephyr-sdk-0.9.1/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: section .intList VMA [0000000020004000,0000000020004093] overlaps section noinit VMA [0000000020002f2c,000000002000456b]
/opt/zephyr-sdk-0.9.1/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: region `SRAM' overflowed by 1388 bytes
collect2: error: ld returned 1 exit status

carlescufi, 2017-06-21:
@cvinayak and myself will take a look tomorrow

carlescufi, 2017-06-22:
https://github.com/zephyrproject-rtos/zephyr/pull/570

nashif, 2017-06-22:
fixed by #570  and merged

closed 2017-06-22 by nashif