zephyrproject-rtos/zephyr #698: drivers/i2c/i2c_ll_stm32_v2.c: error: 'timing' may be used uninitialized in this function

pfalcon, 2017-07-06:
This breaks sanitycheck for disco_l475_iot1, log:

https://ci.linaro.org/view/lite-iot-ci/job/zephyr-upstream/PLATFORM=disco_l475_iot1,ZEPHYR_GCC_VARIANT=zephyr,label=docker-xenial-amd64-13/828/console

Way to reproduce (from log):
~~~
sanitycheck --platform disco_l475_iot1 --inline-logs --build-only --outdir /home/buildslave/srv/zephyr/zephyr/disco_l475_iot1 --no-clean --enable-slow --ccache
~~~

Exact error:
~~~
00:16:42.748 In file included from /home/buildslave/workspace/zephyr-upstream/arch/arm/soc/st_stm32/stm32l4/soc.h:47:0,
00:16:42.748                  from /home/buildslave/workspace/zephyr-upstream/boards/arm/disco_l475_iot1/board.h:10,
00:16:42.748                  from /home/buildslave/workspace/zephyr-upstream/drivers/i2c/i2c_ll_stm32_v2.c:15:
00:16:42.748 /home/buildslave/workspace/zephyr-upstream/drivers/i2c/i2c_ll_stm32_v2.c: In function 'stm32_i2c_configure_timing':
00:16:42.748 /home/buildslave/workspace/zephyr-upstream/ext/hal/st/stm32cube/stm32l4xx/drivers/include/stm32l4xx_ll_i2c.h:902:20: error: 'timing' may be used uninitialized in this function [-Werror=maybe-uninitialized]
00:16:42.748    WRITE_REG(I2Cx->TIMINGR, Timing);
00:16:42.748    ~~~~~~~~~~~~~~~~~^~~~~~~~~~~
00:16:42.748 /home/buildslave/workspace/zephyr-upstream/drivers/i2c/i2c_ll_stm32_v2.c:265:8: note: 'timing' was declared here
00:16:42.748   u32_t timing;
00:16:42.748         ^~~~~~
00:16:42.748 cc1: all warnings being treated as errors
~~~

pfalcon, 2017-07-06:
@erwango , @ldts : Can you please look into this?

pfalcon, 2017-07-06:
Also in https://app.shippable.com/github/zephyrproject-rtos/ci-test/runs/282/2/tests

pfalcon, 2017-07-06:
As clearly related, posting here. https://ci.linaro.org/view/lite-iot-ci/job/zephyr-upstream/825/PLATFORM=stm32_mini_a15,ZEPHYR_GCC_VARIANT=gccarmemb,label=docker-xenial-amd64-13/console

~~~
00:28:09.929 /home/buildslave/workspace/zephyr-upstream/drivers/i2c/i2c_ll_stm32.c:92:12: error: 'i2c_stm32_init' defined but not used [-Werror=unused-function]
00:28:09.929  static int i2c_stm32_init(struct device *dev)
00:28:09.929             ^~~~~~~~~~~~~~
00:28:09.929 /home/buildslave/workspace/zephyr-upstream/drivers/i2c/i2c_ll_stm32.c:87:36: error: 'api_funcs' defined but not used [-Werror=unused-const-variable=]
00:28:09.929  static const struct i2c_driver_api api_funcs = {
00:28:09.929                                     ^~~~~~~~~
00:28:09.929 cc1: all warnings being treated as errors
~~~

This is apparently happens because none of the I2C ports is enabled in i2c_ll_stm32.c

pfalcon, 2017-07-06:
Oops, the latest issue is apparently fixed in the latest master (but not on the level of i2c_ll_stm32.c, so might reoccur for another SoC).

galak, 2017-07-06:
I just summited a fix for this in PR #703

galak, 2017-07-06:
Should now be fixed

closed 2017-07-06 by galak