zephyrproject-rtos/zephyr #1082: build all tests have issues for devices that don't exist on a given board

galak, 2017-08-10:
When we try and do a build only test of all sensor drivers for example we run into issues in that various config options for the sensor driver (mainly the device name) aren't always specified for boards that don't have those sensors.  We should be able to build test such drivers w/o going through a ton of pain.

For now we've worked around this by disabling things: See commit b4598c19e3dea581125aae5f097dfa999f9e28c0f.

Additionally we need some means of doing this testing without having to constantly updating a defconfig/prj.conf file to include the sensors.

nashif, 2017-08-10:
The purpose of that test *is* to test build the sensors and not to test how they build or not on every board. The comment FXOS8700 has its own problem, it was made dependent on a specific board, which is the the origin of the problem (introduced in 563f8f1464525f6b20b57a2090a4c53145a188dd). A sensor should be board independent, a board might have some sensor built in, but we should not make the sensors depends on the board.

nashif, 2018-02-08:
I think this is resolved and had a fix

closed 2018-02-08 by nashif