zephyrproject-rtos/zephyr #1166: Keeping reusable components under samples/ leads to build issues

pfalcon, 2017-08-17:
Steps to reproduce:

1. Try to build samples/net/echo_server with BLE support:
~~~
make BOARD=96b_carbon CONF_FILE=prj_bt.conf
~~~

This will lead to following object layout:
~~~
$ find -name "*.o"
./bluetooth/gatt/ipss.o
./outdir/96b_carbon/arch/built-in.o
./outdir/96b_carbon/arch/common/built-in.o
./outdir/96b_carbon/arch/common/isr_tables.o
./outdir/96b_carbon/arch/arm/built-in.o
~~~

As can be seen, ipss.o goes outside the normal outdir/$(BOARD)/ hierarchy. This of course causes conflicts when building the same samples for other boards/archs (I don't see errors, but there're definitely spurious rebuilds).

Another issue trying to build a sample like samples/net/sockets/echo . As can be seen, this has 4 layers of hierarchy, while samples/net/common/Makefile.common refers to ipss.o as:
~~~
obj-y += ../../../bluetooth/gatt/ipss.o
~~~
so, it simply cannot find it for the samples like that.

pfalcon, 2017-08-17:
@jukkar , @carlescufi, @Vudentz : Please let me know if you have an idea how best to fix this. My idea would be to introduce subsys/bluetooth/service/ , and start with moving ipss.o there, as that's truly reusable service as used in the codebase. Then other services from samples/bluetooth/gatt/ may follow, if needed.

pfalcon, 2017-08-17:
Not that similar issue exists for other subsystems, e.g. recently similar was fixed for net, and this ticket talks about usb cases: https://github.com/zephyrproject-rtos/zephyr/pull/716

carlescufi, 2017-08-17:
Agreed, I've never liked the samples/bluetooth/gatt folder. I do think that (standard) services belong in the subsys folder or, alternatively, in the lib folder. So as I see it either we move those to subsys/bluetooth/host/<gatt,services> or to lib/bluetooth/gatt

pfalcon, 2017-08-17:
@carlescufi : Thanks for quick response. There's currently no lib/bluetooth/ , and the current content of lib/ is rather generic (libc/its adaptations, json). And having lib/bluetooth/ would mean that subsystem is split between 2 subtrees.

On the other hand, taking subsys/net/ as an example, it has net/ip/ for the "core" code, and net/lib/ for various protocols and helpers. So perhaps, keeping everything under subsys/bluetooth/ , in a suitable subdir, is possible.

carlescufi, 2017-08-17:
@pfalcon right, I think subsys/bluetooth/gatt or subsys/bluetooth/services would be the right location for this

nashif, 2017-08-17:
I talked about this with @jhedberg just recently, the idea is to move this to subsys/ as an application layer.

jhedberg, 2017-08-17:
The way I see this code is that it's an extension of the application domain, or in other words a library that the application uses. One requirement of it is that it must only use public APIs. I'm therefore a little reluctant at putting it under subsys/bluetooth since it's not really on the same level as the Bluetooth subsystem, rather a layer above it (i.e. the application). Just like HCI drivers are a layer below the (host) stack, and therefore separately under drivers/bluetooth, I don't see anything wrong with putting this code under lib/bluetooth/

pfalcon, 2017-08-17:
Well, whatever works to resolve the situation. Ideally it would be nice to have some inter-subsystem consistency and patterns, and as I point, net uses subsys/net/lib/ for the very similar purpose (and with very similar constraints, like public API usage), but if for BT, lib/bluetooth/ offers better separation, well...

carlescufi, 2017-08-17:
@jhedberg @pfalcon I would always put consistency before other concerns, so I'd be in favor of either placing it under subsys/ or talking to the net people so that they move theirs to lib/

nashif, 2017-08-17:
#bikeshedding alert: I would refrain from lib/ and follow how it was done in subsys/net, lets keep the code for bluetooth in one place under subsys and create the needed structure under subsys/blueooth

jhedberg, 2017-08-17:
Then I'd probably go for subsys/bluetooth/lib. Anyway, this is a non-issue for IPSP since the existing "lib" for it can be completely removed. @Vudentz will fill in the details.

Vudentz, 2017-08-17:
In case of IPSP there is no such problem anymore since as NET_APP takes care of it, so there must be something pulling it since I removed ipsp.o from Makefile in:

https://github.com/zephyrproject-rtos/zephyr/pull/979

nashif, 2017-08-17:
```
git grep ipss.o
samples/net/common/Makefile.common:obj-y += ../../../bluetooth/gatt/ipss.o
```

Vudentz, 2017-08-17:
https://github.com/zephyrproject-rtos/zephyr/pull/1168

pfalcon, 2017-08-17:
https://github.com/zephyrproject-rtos/zephyr/pull/1168 indeed helped with the immediate issue, thanks! So, I guess the rest can wait until 1.10 window opens.

mbolivar, 2017-08-18:
Seems like a dupe of https://jira.zephyrproject.org/browse/ZEP-2384, which was marked closed, but now I'm not sure why.

Also, are we using GH issues now, not JIRA?

closed 2017-11-14 by nashif