zephyrproject-rtos/zephyr #1181: net/lib/app/client.c appear to have duplicate field assignments

pfalcon, 2017-08-18:
Seeing stuff like:

~~~
        net_sin6(&ctx->ipv6.remote)->sin6_family = AF_INET6;
        ctx->ipv6.remote.family = AF_INET6;
~~~

In the middle of another refactor, recording to not forget.

pfalcon, 2017-08-18:
@jukkar : fyi

jukkar, 2017-08-18:
Those are different fields, so need to set separately. Probably the sin6_family is not really used but as the field is there we should set it.

pfalcon, 2017-08-21:
Yes, and it's mandated by POSIX: http://pubs.opengroup.org/onlinepubs/009696699/basedefs/sys/socket.h.html :

> When a sockaddr_storage structure is cast as a sockaddr structure, the ss_family field of the sockaddr_storage structure shall map onto the sa_family field of the sockaddr structure. When a sockaddr_storage structure is cast as a protocol-specific address structure, the ss_family field shall map onto a field of that structure that is of type sa_family_t and that identifies the protocol's address family.

Actually, for some time I wanted to RFC going even further for Zephyr, and employ the fact that port and address also have the same offset in IPv4 vs IPv6 sockaddr's, now recorded as https://jira.zephyrproject.org/browse/ZEP-2106?focusedCommentId=19680&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-19680 .

nashif, 2017-11-28:
@pfalcon any update on this one? Is this still an issue?

pfalcon, 2017-11-28:
@nashif : Yes. However, this clearly doesn't fit into 1.10, assigned 1.11 milestone.

nashif, 2018-02-07:
@jukkar are you looking into this?

pfalcon, 2018-02-07:
Let me chime in with the idea that @jukkar might have more important things to do in general, and this issue is not too critical for 1.11 (a cleanup, not a bug). So, assigning to myself (keeping @jukkar assigned still as the maintainer, I don't know if that's right thing to do, but I always did), and removing 1.11.

pfalcon, 2018-02-07:
(I usually assign all networking tickets to @jukkar as the maintainer for "triaging", which may include reassigning to other parties. This issue was acked by him, thanks.)

jukkar, 2019-02-21:
net-app has been removed from zephyr

closed 2019-02-21 by jukkar