zephyrproject-rtos/zephyr #1205: C++ usage

drahnr, 2017-08-21:
```
In file included from /home/bernhard/Projects/nello_firmware_ng/src/main.cpp:4:0:
/home/bernhard/Projects/zephyr/include/sensor.h: In function 'int sensor_attr_set(device*, sensor_channel, sensor_attribute, const sensor_value*)':
/home/bernhard/Projects/zephyr/include/sensor.h:271:45: error: invalid conversion from 'const void*' to 'const sensor_driver_api*' [-fpermissive]
  const struct sensor_driver_api *api = dev->driver_api;
                                        ~~~~~^~~~~~~~~~
/home/bernhard/Projects/zephyr/include/sensor.h: In function 'int sensor_trigger_set(device*, sensor_trigger*, sensor_trigger_handler_t)':
/home/bernhard/Projects/zephyr/include/sensor.h:299:45: error: invalid conversion from 'const void*' to 'const sensor_driver_api*' [-fpermissive]
  const struct sensor_driver_api *api = dev->driver_api;
```

my config includes the `CONFIG_CPLUSPLUS=y` the target is the nrf52 dk dev boad.

How do I get around this? Specifiying `-fpermissive` does not solve my issue. I tried to include all C includes in `extern "C"` but no success.

drahnr, 2017-08-21:
This happens with 1.8.1 and the latest git of today

drahnr, 2017-08-21:
`CXXFLAGS=-fpermissive make` works though, but specifying it inside the makefile does not:

```
#required since pgio.h and snesors.h of zephyr otherwise fail to build
CXXFLAGS="-fpermissive"

BOARD ?= nrf52_pca10040
CONF_FILE = prj.conf

include ${ZEPHYR_BASE}/Makefile.inc
```

szundi, 2019-05-30:
This is an issue for me too, i have to use -fpermissive, with that it works, but... it's permissive :)

commit 32e3756d (2019-12-12): include: Fixes #1205, C++ usage of sensor.h

I ran into issue #1205 earlier today and realized the fix was to simply
provide the proper casts.  The issue is that C++ is less permissive than
C here, erroring when trying to implicitly convert from `void *` to
`struct gpio_driver_api *`.  The same cast is done in
include/drivers/gpio.h, which is why I did that here as well.

This fix was validated by compiling my C++ application successfully and
also successfully running my app on my board, interacting with sensors.
closed 2019-12-12 by MaureenHelm