zephyrproject-rtos/zephyr #1214: uint32_t type changed while compiling with minimal libc and newlibc

ehudmarvell, 2017-08-22:
Hi,

I wrote the following simple code:
and compile it in both cases: while using minimal libc and newlibc
```
void main(void)
{
	uint32_t i = 1;
	printf("Hello!, i = %d\n", i);
}
```

while compiling with minimal libc uint32_t type is "**unsigned int**"
while compiling with new libc uint32_t type  is "**long unsigned int**"

so basically the code won't compile in both cases withou error, because "minimal libc" use **%d** and newlibc use **%ld**.

Please Advise

Thanks a lot,
Ehud

tbursztyka, 2017-08-22:
Hi Ehud,

You have to use Zephyr's own type, u32_t, u16_t, u8_t for the unsigned, 
s32_t ...etc  for the signed ones.
See includes/zephyr/types.h

And for more information: commit id 789081673f

These types were introduced for that reason (libc differences), among 
others.

Btw, %u should be used for an unsigned. %d is for signed integer.

Tomasz

ehudmarvell, 2017-08-22:
Thanks, I will try

closed 2017-08-22 by ehudmarvell