zephyrproject-rtos/zephyr #1218: tests/benchmarks/timing_info/testcase.yaml#test failed @ nrf52_pca10040

inakypg, 2017-08-22:
eval failed: found expected (for failure) `(USAGE FAULT|[Ff]atal fault in )` in console `jfsotc03/nrf52-37:default` at 1.62s
console output: ***** delaying boot 1000ms (per build configuration) *****
console output: ***** MPU FAULT *****
console output:   Executing thread ID (thread): 0x200010a4
console output:   Faulting instruction address:  0x200024a8
console output:   Instruction Access Violation
console output: Fatal fault in ISR! Spinning...

gitrev 4e439653aaaf0c2ab8a4333cebb1e1385905354a

nashif, 2017-08-23:
@carlescufi  FYI

youvedeep, 2017-08-23:
Further there is another thread at GitHub where discussion about this is tracked.

youvedeep, 2017-08-23:
https://github.com/zephyrproject-rtos/zephyr/pull/1095

carlescufi, 2017-08-23:
@nashif I am following up in the #1095 thread, thanks!

AdithyaBaglody, 2017-08-23:
I feel the issue is with the AAPCS for ARM. if the scratch registers are saved it should be okay.

nashif, 2017-08-31:
fixed by #1095

closed 2017-08-31 by nashif