zephyrproject-rtos/zephyr #1240: tests/net/mld/ failure on qc1000:x86

inakypg, 2017-08-24:
toolchain: zephyr
Zephyr: caa377943a809ceffd40eef04a9f21a883e7f9cc

```
***** delaying boot 1000ms (per build configuration) *****
***** BOOTING ZEPHYR OS v1.9.0-rc1 (delayed boot 1000ms) - BUILD: Aug 23 2017 23:34:35 *****
Running test suite net_mld_test
===================================================================
starting test - mld_setup
PASS - mld_setup.
===================================================================
starting test - join_group

    Assertion failed at /home/jenkins/workspace/zephyr-master-tcf-v0.11-branch/LABEL/verify/SHARD/3-3/ZEPHYR_GCC_VARIANT/zephyr/target.git/tests/net/mld/src/main.c:203: join_group: Cannot join IPv6 multicast group (ret not equal to 0)
FAIL - join_group.
===================================================================
starting test - leave_group
PASS - leave_group.
===================================================================
starting test - catch_join_group
PASS - catch_join_group.
===================================================================
starting test - catch_leave_group
PASS - catch_leave_group.
===================================================================
starting test - verify_join_group
PASS - verify_join_group.
===================================================================
starting test - verify_leave_group
PASS - verify_leave_group.
===================================================================
starting test - catch_query
PASS - catch_query.
===================================================================
starting test - verify_send_report

    Assertion failed at /home/jenkins/workspace/zephyr-master-tcf-v0.11-branch/LABEL/verify/SHARD/3-3/ZEPHYR_GCC_VARIANT/zephyr/target.git/tests/net/mld/src/main.c:203: join_group: Cannot join IPv6 multicast group (ret not equal to 0)
FAIL - verify_send_report.
===================================================================
starting test - test_allnodes
PASS - test_allnodes.
===================================================================
starting test - test_solicit_node
PASS - test_solicit_node.
===================================================================
===================================================================
RunID: ci-170823-1622-32:zwxw
PROJECT EXECUTION FAILED

```

inakypg, 2017-08-24:
Also happens on qemu_x86

nashif, 2017-08-24:
> Also happens on qemu_x86

This is run by sanitycheck on regular basis, so not sure why you see a failure.

PR #1245 (2017-08-26): tests: net: mld: Ignore already joined error when sending report
The verify_send_report() failed in quark_se_c1000_devboard because
the multicast group was already joined. We can safely ignore this
error for this specific test.

Fixes #1240
commit e0dd4ff7 (2017-08-26): tests: net: mld: Ignore already joined error when sending report

The verify_send_report() failed in quark_se_c1000_devboard because
the multicast group was already joined. We can safely ignore this
error for this specific test.

Fixes #1240
closed 2017-08-26 by nashif