zephyrproject-rtos/zephyr #1241: tests/net/ipv6/ FAILED on qc1000:x86

inakypg, 2017-08-24:
toolchain: zephyr
Zephyr: caa377943a809ceffd40eef04a9f21a883e7f9cc

```
***** delaying boot 1000ms (per build configuration) *****
***** BOOTING ZEPHYR OS v1.9.0-rc1 (delayed boot 1000ms) - BUILD: Aug 23 2017 23:39:00 *****
starting test - test init
FAIL - test_init@286. Cannot add multicast IPv6 address ff02::1
failed
starting test - IPv6 compare prefix
passed
starting test - IPv6 neighbor lookup fail
passed
starting test - IPv6 add neighbor
passed
starting test - IPv6 neighbor lookup ok
passed
starting test - IPv6 send NS extra options
passed
starting test - IPv6 send NS no options
passed
starting test - IPv6 handle RA message
passed
starting test - IPv6 parse Hop-By-Hop Option
passed
starting test - IPv6 change ll address
passed
starting test - IPv6 prefix timeout
passed
===================================================================
RunID: ci-170823-1622-32:fhy4
PROJECT EXECUTION FAILED

```

inakypg, 2017-08-24:
Suspected similar to #1240 . Also fails on qemu_x86 and arduino_101

jukkar, 2017-08-24:
Works ok for me with qemu and quark_se. I do not have a101 to try.

inakypg, 2017-10-23:
This seems a dup of #4401

closed 2017-10-30 by linkmeyer