zephyrproject-rtos/zephyr #1242: tests/kernel/mutex/mutex/ FAILED @ esp32

inakypg, 2017-08-24:
toolchain: espressif
Zephyr: caa377943a809ceffd40eef04a9f21a883e7f9cc

```
***** BOOTING ZEPHYR OS v1.9.0-rc1 (delayed boot 1000ms) - BUILD: Aug 23 2017 19:23:27 *****
Done LOCKING!  Current priority = 5
Testing recursive locking
Recursive locking tests successful
PASS - regression_task.
===================================================================
===================================================================
RunID: ci-170823-1622-32:i0kn
PROJECT EXECUTION SUCCESSFUL
*** Unhandled exception ****
Exception cause 0 (illegal instruction):
  EPC1     : 0x4008281e EXCSAVE1 : 0x3d3d3d3d EXCVADDR : 0x00000000
Program state (PS):
  INTLEVEL : 03 EXCM    : 0 UM  : 1 RING : 0 WOE : 1
  OWB      : 04 CALLINC : 2
Current thread ID = 0x3ffb1698
Faulting instruction address = 0xdeaddead
Fatal fault in essential thread! Spinning...

```

lpereira, 2017-08-24:
I can't reproduce this on caa3779: I get the "PROJECT EXECUTION SUCCESSFUL" message, and no exceptions are thrown after that.  Are you getting the crash immediately after the test ends, or does it take some time for it to happen?  I've let it running here over a minute and nothing happened.

lpereira, 2017-08-24:
Can't reproduce on 83088a23 (current HEAD at the moment) either.

nashif, 2017-08-24:
I also was not able to reproduce

inakypg, 2017-08-24:
It's hard to tell, but it is happening consistently on the automation results. My first hunch would be it'd be an artifact of the power down process -- but then it'd affect other testcases, and it does not.

One thing I missed is esp-ipf is v2.0-rc1-956-g81f98c0

inakypg, 2017-08-24:
Leandro: timing wise, it is hard/er to know; if you guys can't reproduce, let's let it run a few more times--maybe it got fixed by something else.

lpereira, 2017-10-23:
Closing as this cannot be reproduced on current HEAD.

closed 2017-10-23 by lpereira