zephyrproject-rtos/zephyr #1266: cmake: qemu_x86 with MMU enabled not working

nashif, 2017-08-26:
running qemu_x86 with MMU enabled does not work as expected. If we disable MMU it boots up in Qemu just fine.

SebastianBoe, 2017-08-28:
I have been debugging this for a full day already, so I will take this one.
Did not know of the MMU workaround.

SebastianBoe, 2017-08-28:
Hmm, it is not the generation of mmu_tables.o that is wrong at least.

mmu_tables.o is identical with KBuild.
CMake generates an elf where every symbol has identical size with the KBuild as well,
so presumably all symbols are equivalent, but located at different addresses.

Working theory is that MMU-builds require certain symbols to be placed at certain 
places, either absolute, or relative to some other symbols. And that this hasn't been
ported correctly to CMake.

nashif, 2017-08-28:
@andrewboie any hints of known caveats you can provide here? This is related to PR #1107

andrewboie, 2017-08-28:
I haven't studied the CMake patch in detail, however:

1) arch/x86/Makefile.mmu needs to be ported to CMake correctly
2) I assume we are still using the same linker scripts? include/arch/x86/linker.ld

> Working theory is that MMU-builds require certain symbols to be placed at certain
places, either absolute, or relative to some other symbols. And that this hasn't been
ported correctly to CMake.

Yes, it does. But this is all taken care of in x86's linker.ld, are we not using it?

SebastianBoe, 2017-08-29:
> arch/x86/Makefile.mmu needs to be ported to CMake correctly
AFAICT it has been. Makefile.mmu generates mmu_tables.o, and I have verified that mmu_tables.o is identical between KBuild and CMake.

> I assume we are still using the same linker scripts? include/arch/x86/linker.ld
Correct. And it has been verified that the preprocessed linker script is identical.

In GDB I see that emu "crashes" on this line, going to find out why now ...
https://github.com/SebastianBoe/zephyr/blob/cmake/arch/x86/core/crt0.S#L276

SebastianBoe, 2017-08-29:
Seems it is X86_STACK_PROTECTION that is causing qemu_x86 to crash, not MMU.

qemu crashes in CONFIG_X86_STACK_PROTECTION ifdef'd code and qemu_x86 will run
with a prj.conf like this:

CONFIG_X86_MMU=y
CONFIG_X86_STACK_PROTECTION=n

SebastianBoe, 2017-08-29:
Ah, think I found the bug now ...

SebastianBoe, 2017-08-29:
Fixed, root cause was Makefile.gdt not being ported correctly.

https://github.com/zephyrproject-rtos/zephyr/pull/1107/commits/39f66dfe3b916d74db6597f1ab7ca6a5559c729c

nashif, 2017-08-29:
fixed in cmake PR

closed 2017-08-29 by nashif