zephyrproject-rtos/zephyr #1270: Issue : Information CC3220SF LaunchXL

IotPanorama, 2017-08-28:
There is a wrong information on Zephyr project site about CC3220SF launchpad. It has 4 MB flash where as the document in the link below says it has 1MB flash.  [https://www.zephyrproject.org/doc/boards/arm/cc3220sf_launchxl/doc/cc3220sf_launchxl.html] .

ydamigos, 2017-08-28:
Could you please provide a link where it mentions that CC3220SF launchpad has 4 MB flash?
TI's product page (http://www.ti.com/tool/CC3220SF-LAUNCHXL) mentions 1MB Flash.

dbkinder, 2017-08-28:
I see lots of references to the board only having 1MB flash:  https://store.ti.com/CC3220SF-LAUNCHXL-SimpleLink-Wi-Fi-CC3220SF-Wireless-Microcontroller-LaunchPad-Development-Kit-P50638.aspx https://www.digikey.com/en/product-highlight/t/texas-instruments/cc3220sf-launchpad http://www.ti.com/lit/ds/symlink/cc3220.pdf

GAnthony, 2017-08-28:
@IotPanorama, please provide reference showing the LaunchXL may have 4 Mb internal flash.

All TI CC3220SF documents I've seen, plus the LaunchXL description itself from digikey, have boards with 1Mb flash available:
https://www.digikey.com/en/product-highlight/t/texas-instruments/cc3220sf-launchpad

IotPanorama, 2017-11-23:
@GAnthony , sorry for being late, CC3220SF have a 1Mb xip flash internally but the CC3220SF Launchxl  has a 32Mbit (4MB) flash (MX25R3235F) on board. Each CC3220x device need that external flash. you can find the information in block diagram present in this Document( http://www.ti.com/lit/ug/swru463/swru463.pdf )

GAnthony, 2017-12-05:
@IotPanorama, you are correct.  I verified the *external* flash is 4MB using the TI UniFlash tool.  Internal flash is still 1MB.  I'll make this more clear in the documentation.

PR #5264 (2017-12-05): boards: arm: cc3220sf_launchxl: doc: clarify external flash size
External serial flash on this board is 4MB.

Previously, this was confused with internal flash, which is 1MB.

Fixes #1270
commit 60d16bda (2017-12-05): boards: arm: cc3220sf_launchxl: doc: clarify external flash size

External serial flash on this board is 4MB.

Previously, this was confused with internal flash, which is 1MB.

Fixes #1270
closed 2017-12-05 by galak