zephyrproject-rtos/zephyr #1281: stm32_ll_spi driver does not support stm32f1soc

chunshanzhu, 2017-08-29:


guo-mkcooc, 2017-10-27:
I also encountered the same problem.

I plan to use spi with the stm32f107 , but dts or pinmux seem can't configure SPI pins. 
however, stm32f4, f3, l4 all works with pinmux except f1.

mgorzkowski, 2017-12-13:
I have the same problem.
I want to use SPI on stm32_min_dev board (stm32f103)

chunshanzhu, 2017-12-13:
stm32f1 does not have the SPI_SR_FRE regesiter bit，and not support the LL_SPI_PROTOCOL_MOTOROLA protocol.

ydamigos, 2017-12-13:
I will check it.

ydamigos, 2017-12-14:
Just a quick update.
I started working on the issue at https://github.com/ydamigos/zephyr/tree/stm32f1-spi. 
The driver compiles fine but when I use the spi-loopback test, I get a bus fault, pointing to line 245 of spi_contex.h. The error isn't specifically in this part of the code. I will try to find the root of it.
There is a similar issue #3850 for stm32f3 family.

ydamigos, 2017-12-15:
Could you test PR #5413?

PR #5413 (2018-02-01): drivers: spi_ll_stm32: Enable SPI driver for F1 family
Enables SPI driver for STM32F1 SoCs.

Tested on olimexino_stm32 using spi_loopback test.

Fixes #1281 and #3850.
closed 2018-02-01 by galak