zephyrproject-rtos/zephyr #1289: C++ 11 support!

soleilos, 2017-08-29:
Hi,

I am quite new to zephyr and would like to know if the C++ 11 is supported?

Any thoughts on this is appreciated.

Thank you.

ThomasGrausgruber, 2017-09-07:
Hi!

I am also interested in this topic. Looking forward to a concrete plan for C++ 11 support 👍

samirame, 2017-09-08:
Hi Zephyr,

i also find the c++ support an outstanding feature to compete other RTOSs and to enrich the Zephyr ecosystem with more applications supports.

Any plan for this would be great.

mbolivar, 2017-10-25:
Looks like this issue and https://github.com/zephyrproject-rtos/zephyr/issues/4028 are dupes. The discussion in the other issue is further along, so maybe close this one as the dupe, even though it's a bit older?

Tenderson, 2017-10-26:
@mbolivar, thank you for the notice. I close this issue, however I appreciate if you have some thoughts to share on the issue #4028.

closed 2017-10-26 by Tenderson