zephyrproject-rtos/zephyr #1291: Posix PSE51 Support

nashif, 2017-08-30:
- [ ] 1003.1b-93 Realtime Signals
- [ ] 1003.1b-93 Semaphores
- [ ]  1003.1b-93 Hi Res. Clocks & Timers
- [ ]  1003.1b-93 Shared Memory
- [ ] 1003.1b-93 IPC Message Passing
- [ ] 1003.1c-95 Threads
- [ ] 1003.1c-95 Thread Safe Functions
- [ ] 1003.1c-95 Thread Attribute Stack Address
- [ ] 1003.1c-95 Thread Attribute Stack Size
- [ ]  1003.1c-95 Thread Priority Scheduling
- [ ] 1003.1c-95 Thread Priority Inheritance
- [ ] 1003.1c-95 Thread Priority Protection

LorenzMeier, 2018-02-20:
@nashif Could you share the roadmap for this or a planning document?

ceolin, 2018-10-30:
@nashif looking the features, aren't we targeting pse51 ? pse52 is pse51 + file systems + asynchronous I/O.

If we're targeting pse52 we need to add:
 - memory mapped files
 - files and directories
 - Asynchronous I/O