zephyrproject-rtos/zephyr #1324: Use correct barriers instead of relying on compiler optimizations

agross-linaro, 2017-08-31:
The gen_isr_table kernel test uses GCC optimization of O0 to ensure that proper execution is achieved.  Instead of using optimizations, it should be using a proper barrier mechanism.

The optimization change is located in:
 tests/kernel/gen_isr_table/prj.conf

Need to add in an __ISB() or whatever barrier is required and test to make sure the execution is correct.

bryan-hunt, 2017-09-14:
Also relating to ARM: arch/arm/core/cpu_idle.S makes WFI & WFE calls without DSB beforehand.

d3zd3z, 2019-12-20:
A GCC optimization of O0 is only going to be equivalent to a compiler barrier (implemented by any `asm volatile` with `"memory"` in the clobber argument). This only serves to prevent memory reordering by the compiler. We should figure out the necessary CPU barriers to make this correct at that level.

ioannisg, 2021-02-10:
> Also relating to ARM: arch/arm/core/cpu_idle.S makes WFI & WFE calls without DSB beforehand.
> 
> http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dai0321a/BIHICBGB.html

@bryan-hunt this has been fixed for quite some time now 👍

ioannisg, 2021-04-23:
This is solved for Cortex-M, not sure if we would like to keep this open, to get fixes for other ARCHes

kartben, 2023-08-20:
I believe this has been addressed a while ago. Plus, there's even a barrier API now #57439

closed 2023-08-20 by kartben