zephyrproject-rtos/zephyr #1350: Provide build type in version.h

carlescufi, 2017-09-04:
We need a 2-bit identifier of the build type:

* Production release
* Release candidate
* Development build

So we can propagate that up via HCI to the Host

carlescufi, 2017-09-04:
Related to https://github.com/zephyrproject-rtos/zephyr/issues/1333

nashif, 2017-09-04:
we have been using EXTRAVERSION for the rc, like we it right now for 1.9, development releases were set as 1.x.99

carlescufi, 2017-09-04:
@nashif but that is not quite equivalent to a fully-fledged build type, in case the patch level goes > 99. 

Otherwise I'm forced to do things like:

```
#if KERNEL_PATCHLEVEL == 99
..
#endif
```
Ideally we'd have something like:

`KERNEL_BUILD_TYPE 0x01`

carlescufi, 2017-09-04:
@holtmann feel free to add your thoughts

pfalcon, 2017-09-04:
>  in case the patch level goes > 99.

Not too late to change it to 999 ;-).

> Otherwise I'm forced to do things like:
> #if KERNEL_PATCHLEVEL == 99
> ..
> #endif

I guess the talk is who'll be (not forgetting to) updating all these bits and pieces. Otherwise, we can have

~~~
#if KERNEL_PATCHLEVEL == 99
#define KERNEL_BUILD_TYPE KERNEL_BUILD_TYPE_DEVEL
#else  // checking EXTRAVERSION may be tricky, yeah
#endif
~~~

without need to introduce for knobs.

carlescufi, 2017-09-04:
@pfalcon So my idea was to find that sorted out already in version.h, a macro that is part of kernel.h and that tells you which kind of build that is:

In kernel.h:

```
#define KERNEL_BUILD_DEV 0x00
#define KERNEL_BUILD_RC 0x01
#define KERNEL_BUILD_PROD 0x02
```

In version.h:

```
#define KERNEL_BUILD_TYPE KERNEL_BUILD_PROD
```

nashif, 2017-09-05:
i am not saying the .99 solution is an ideal one, it is what we have an we need to find a better way for managing this.
We should also consider reserving a version header in the image that can be injected with versions and build-ids  post-build as well. 
Note that the version handling is going to change when we move to cmake, meaning it will be done using some other method, so I would wait and implement all of this in the context of cmake.

marc-hb, 2019-04-30:
@carlescufi , would PR #9188 help?

carlescufi, 2019-05-03:
@marc-hb yes, it probably would since we'd be able to retrieve the build type from that structure that the PR adds. Thinking of picking up that PR?

carlescufi, 2019-05-03:
@marc-hb in particular I wanted to use this to include this info in the string that is returned by a zephyr-based Bluetooth controller using the following Vendor-Specific command:

https://github.com/zephyrproject-rtos/zephyr/blob/master/doc/reference/bluetooth/hci.txt#L499

Which is current implemented as:

https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/hci/hci.c#L1870

marc-hb, 2019-05-06:
No I don't have any need or time to resume the ELF PR #9188, sorry. This being said, I have a modest contribution: see my latest comment in PR #1333 and help review comments I'm adding in new PR #15950

closed 2022-11-17 by nashif