zephyrproject-rtos/zephyr #1381: HMC5883L driver config error

longcongduoi, 2017-09-06:
Data unchanged after read

	/* configure device */
	chip_cfg[0] = idx << HMC5883L_ODR_SHIFT;
	chip_cfg[1] = drv_data->gain_idx << HMC5883L_GAIN_SHIFT;
	chip_cfg[2] = HMC5883L_MODE_CONTINUOUS;

	if (i2c_burst_write(drv_data->i2c, HMC5883L_I2C_ADDR,
			    HMC5883L_REG_CONFIG_A, chip_cfg, 3) < 0) {
		SYS_LOG_ERR("Failed to configure chip.");
		return -EIO;
	}

I' ve changed to 

	/* configure device - FIXME */  
	chip_cfg[0] = idx << HMC5883L_ODR_SHIFT;
	chip_cfg[1] = drv_data->gain_idx << HMC5883L_GAIN_SHIFT;
	chip_cfg[2] = HMC5883L_MODE_CONTINUOUS;

	if (i2c_burst_write(drv_data->i2c, HMC5883L_I2C_ADDR,
			    HMC5883L_REG_CONFIG_A, &chip_cfg[0], 1) < 0) {
		SYS_LOG_ERR("Failed to configure chip.");
		return -EIO;
	}
	
	if (i2c_burst_write(drv_data->i2c, HMC5883L_I2C_ADDR,
			    HMC5883L_REG_CONFIG_B, &chip_cfg[1], 1) < 0) {
		SYS_LOG_ERR("Failed to configure chip.");
		return -EIO;
	}
	
	if (i2c_burst_write(drv_data->i2c, HMC5883L_I2C_ADDR,
			    HMC5883L_REG_MODE, &chip_cfg[2], 1) < 0) {
		SYS_LOG_ERR("Failed to configure chip.");
		return -EIO;
	}

It's ok

nashif, 2017-09-07:
can you submit a PR?

nashif, 2019-02-19:
old issue, closing.

closed 2019-02-19 by nashif