zephyrproject-rtos/zephyr #1392: No module named 'elftools'

spring-z, 2017-09-07:
test build macOS with gcc-arm-none-eabi

make BOARD=stm32f469i_disco

  CHK     include/generated/generated_dts_board.conf
  Using /Users/wenchao/zephyr as source for kernel
  GEN     ./Makefile
  CHK     include/generated/version.h
  CHK     include/generated/generated_dts_board.h
  CHK     misc/generated/configs.c
  CHK     include/generated/offsets.h
Traceback (most recent call last):
  File "/Users/wenchao/zephyr/scripts/gen_offset_header.py", line 8, in <module>
    from elftools.elf.elffile import ELFFile
ModuleNotFoundError: No module named 'elftools'
make[3]: *** [include/generated/offsets.h] Error 1
make[2]: *** [prepare] Error 2
make[1]: *** [sub-make] Error 2
make: *** [all] Error 2

From master branch

spring-z, 2017-09-07:
pip install pyelftools

spring-z, 2017-09-07:
please update wiki and close issues

nashif, 2017-09-07:
what wiki were you looking at? This is documented already here: https://www.zephyrproject.org/doc/getting_started/getting_started.html

spring-z, 2017-09-07:
wiki no add pyelftools

chunshanzhu, 2017-09-07:
pip3 install -r scripts/requiements.txt，software requied to build zephyr list in requiements.txt

nashif, 2017-09-08:
> wiki no add pyelftools

what wiki are you referring to?

spring-z, 2017-09-08:
sorry，i read doc，1.9.0 no release。I mistakenly read the 1.8.0 doc

closed 2017-09-08 by spring-z

dwj-scut, 2019-04-20:
pip3 install -r scripts/requiements.txt

patrick-dojofive, 2021-06-04:
The getting started page linked above DOES NOT even mention elftools

rustand2, 2022-04-29:
It does not help to have pyelftools.

pip install pyelftools
Requirement already satisfied: pyelftools in /home/USERNAME/zephyr/venv/lib/python3.10/site-packages (0.28)

make -j
[  1%] Built target version_h
[  3%] Built target parse_syscalls_target
[  6%] Built target syscall_list_h_target
[  6%] Generating include/generated/driver-validation.h
[  6%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h, include/generated/otype-to-size.h
Traceback (most recent call last):
Traceback (most recent call last):
  File "/home/USERNAME/zephyr/zephyr/scripts/gen_kobject_list.py", line 62, in <module>
  File "/home/USERNAME/zephyr/zephyr/scripts/gen_kobject_list.py", line 62, in <module>
    import elftools
    import elftools
ModuleNotFoundError: No module named 'elftools'
ModuleNotFoundError: No module named 'elftools'
make[2]: *** [zephyr/CMakeFiles/driver_validation_h_target.dir/build.make:73: zephyr/include/generated/driver-validation.h] Error 1
make[2]: *** [zephyr/CMakeFiles/kobj_types_h_target.dir/build.make:74: zephyr/include/generated/kobj-types-enum.h] Error 1
make[1]: *** [CMakeFiles/Makefile2:1489: zephyr/CMakeFiles/driver_validation_h_target.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1515: zephyr/CMakeFiles/kobj_types_h_target.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

rustand2, 2022-04-29:
I actually had to install it outside of the virtualenv also for make to find it. This is obviously not how things are supposed to work

unixbigot, 2022-08-08:
I had this problem just now (mid 2022) on a clean MacOS install.   It turned out to be because I had two copies of west present, one of which explicitly loaded the system python in /Library/Developer/CommandLineTools and one of which used /usr/local/the/actual/python/I/wanted.   Check the output of "head `which west`" and see what python west is running with.

My solution was to get rid of the broken west in my path.

mfiumara, 2022-11-06:
Thanks @unixbigot I had a similar problem, was running inside a virtual environment and had west installed in the system causing this. I uninstalled west from my system and things worked again.

spectrum70, 2024-01-05:
Hi, having same issue on a fresh installation on today, begin of 2024.

west --version
West version: v1.2.0

ModuleNotFoundError: No module named 'elftools'

I have a single copy of west installed.
and nothing changed by "pip3 install pyelftools"

hidsh, 2024-01-15:
My M1 Mac also has a same issue but solved simply `pip install pyelftools` outside the venv.

usmanimtiaz63, 2024-01-15:
I am using Rocky Linux 8.
The documentation, by default (choosing the option locally), suggests to run `pip install -r ~/zephyrproject/zephyr/scripts/requirements.txt` inside virtual environment.
https://docs.zephyrproject.org/latest/develop/getting_started/index.html

While the `pyelftools` don't work from here and needs to be installed globally. May be my environment issue? however worked when I installed it globally using `pip3 install pyelftools` outside virtenv.

LumingLin, 2024-08-28:
I am using Mac M2 and had the same issue. After `pip3 install pyelftools`, I can successfully `python3` and `import elftools`. However, it still shows the same error when I build.

Solution: Look at the error message, you'll see a path like: "/Users/<your_user_name>/.local/pipx/venvs/west/bin/python"
This is the Python interpreter west is using. So we need to Install pyelftools for this specific Python environment:    `/Users/<your_user_name>/.local/pipx/venvs/west/bin/python -m pip install pyelftools` 

This helps me solve the problem.

Azel4231, 2024-10-24:
This is the solution when using pipx, thanks a ton!
Not a python dev, so was very confusing. Seems to me like a weird interaction between package managers and python libraries. I use Ubuntu so pip no longer works ("This environment is externally managed"), so you have to use pipx. But pipx creates a separate python virtual environment (venv) for each lib/tool, like west and qmk. So elftools needs to be installed IN THAT venv as well, hence the above instructions.
The confusing part is that when I activate a venv (with elftools installed) and then call west, somehow the pipx/west venv gets used instead of the one I activated.