zephyrproject-rtos/zephyr #1397: no serialport output

spring-z, 2017-09-07:
hello_word samples run 

`make BOARD=nucleo_f103rb flash`

serialport no output Hello World! arm.

Can you tell me what did I do wrong?

chunshanzhu, 2017-09-07:
port1 or port2? f103rb use port2 as the default serial port

spring-z, 2017-09-07:
ok，I try again

dbkinder, 2017-09-07:
Alas, there is no documentation for the F103RB board, as noted in [ZEP-1907](https://jira.zephyrproject.org/browse/ZEP-1907).  The documentation for the  [F103RE board](https://www.zephyrproject.org/doc/boards/arm/nucleo_f401re/doc/nucleof401re.html) states that, "Nucleo F401RE board has 3 UARTs. The Zephyr console output is assigned to UART2. Default settings are 115200 8N1."

nashif, 2017-09-08:
works fine for me:

Board appears as
```
/sys/bus/usb/devices/3-10.4: STMicroelectronics STM32 STLink serial: '066EFF535550755187031831', tty(s): ttyACM0
```
run:
```
sudo  screen -L  /dev/ttyACM0 115200
```

and you get serial output.

spring-z, 2017-09-08:
no i not get Hello World! arm，

closed 2017-09-08 by spring-z