zephyrproject-rtos/zephyr #1415: Problem with forcing new line in generated documentation.

m-kru, 2017-09-08:
Look at `include/spi.h` lines 149-156. They look good in header file however when it is generated to html it is all in one line, it is not clear and hard to read. doxygen supports @n special command that forces new line in generated documentation. I have added it at the beginning of each line however without any effect on generated documentation text.

nashif, 2017-09-08:
@dbkinder  can you take a look?

dbkinder, 2017-09-08:
Looks like the command to force a newline/linebreak is \n not @n  (according to https://www.stack.nl/~dimitri/doxygen/manual/commands.html)

m-kru, 2017-09-08:
@dbkinder Doxygen manual says: 

> All commands in the documentation start with a backslash (\\) or an at-sign (@). If you prefer you can replace all commands starting with a backslash below by their counterparts that start with an at-sign.

I have check both options `\n` and `@n`,  none of them works.

dbkinder, 2017-09-08:
Yep... further investigation shows this is a known problem with breathe.  I'm checking out the latest version (4.7.2) to see if we can switch to that.

nashif, 2017-09-08:
i have seen some people working around this by adding <br/> in doxygen :-)

dbkinder, 2017-09-08:
using ``@n`` or ``\n`` or ``<br/>`` all generate the same doxygen xml output that's read by sphinx/breathe, and looks like breathe doesn't honor the <linebreak/> xml tag.

So, there is another way to fix this by using ``@param`` to identify the struct members being documented, and using 4 leading spaces when you want to create a ``pre`` block for the part of the struct documentation that describes the bit field layout (and is using spaces to align stuff).  I'm submitting a PR to tweak spi.h with this change.

But there's another problem here: there are two include files defining the spi interface, spi.h and spi_legacy.h with structs and such with the same names, and doxygen is trying to mash them together.

dbkinder, 2017-09-09:
P.S.  You can use a blank line to force a new paragraph.

m-kru, 2017-09-09:
Then documentation in header file will look long and weird.

dbkinder, 2017-09-09:
@m-kru I tried to assign you as a reviewer for https://github.com/zephyrproject-rtos/zephyr/pull/1424 but you don't show up.  Take a look.

closed 2017-09-11 by nashif

ChiShouWu, 2021-03-25:
I found the solution.
If your comment like this, will not work

```
   /**
   * line1 \n
   * line2 \n
   */
```

solution:

``` 
/**
 line1 \n
 line2 \n
*/

This will work

PR #1424 (2017-09-11): doc: improve spi.h doxygen comments layout
Doxygen comments for documenting structs have (known) issues,
and the Breathe addon for Sphinx used to create our API docs
has a known issue with forcing line breaks with @n or `<br/>`

This patch tweaks the comments to use a method used in i2s.h
to use @param comments for the members of a struct, and using
4 leading spaces (as done in i2s.h as well) to create a pre
block for the bit-field layout comments.

Fixes: #1415
commit d74f97d0 (2017-09-11): doc: improve spi.h doxygen comments layout

Doxygen comments for documenting structs have (known) issues,
and the Breathe addon for Sphinx used to create our API docs
has a known issue with forcing line breaks with @n or <br/>

This patch tweaks the comments to use a method used in i2s.h
to use @param comments for the members of a struct, and using
4 leading spaces (as done in i2s.h as well) to create a pre
block for the bit-field layout comments.

Fixes: #1415