zephyrproject-rtos/zephyr #1416: BUG: Regression detected on K64F after commit cd35742a (net/ethernet/arp: Let ethernet L2 managing pkt's reference while sending)

mike-scott, 2017-09-08:
Our upstream CI testing showed the following BUS FAULT after commit cd35742a was introduced:

PROJECT EXECUTION SUCCESSFUL
***** BUS FAULT *****
  Executing thread ID (thread): 0x20002ca8
  Faulting instruction address:  0x24a0a
  Precise data bus error
  Address: 0xbd990258
Fatal fault in essential thread! Spinning...

I will add more details once I have time to debug.

mike-scott, 2017-09-08:
NOTE This is only on master branch and does not affect v1.9

mike-scott, 2017-09-08:
Faulting instruction address correlates to: include/net/buf.h:1024

mike-scott, 2017-09-08:
And verified that reverting commit cd35742a2bae38a468d9163ee797f9e9b5fa194d fixes the BUS FAULT

pfalcon, 2017-09-12:
I built echo_server for frdm_k64f. It starts up ok, but on first ping received, crashes:

~~~
[dev/eth_mcux] [DBG] eth_0_init: MAC 00:04:9f:54:59:16
shell> [dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
***** BUS FAULT *****
  Executing thread ID (thread): 0x20001b28
  Faulting instruction address:  0x44ce
  Precise data bus error
  Address: 0xc63cfe56
Fatal fault in essential thread! Spinning...
~~~

pfalcon, 2017-09-12:
@tbursztyka : ^^^

(also pinged on IRC)

pfalcon, 2017-09-12:
Confirmed https://github.com/zephyrproject-rtos/zephyr/pull/1455 fixes this.

pfalcon, 2017-09-12:
Also, noticed frdm_k64f didn't respond to "ping6 2001:db8::1", confirmed https://github.com/zephyrproject-rtos/zephyr/pull/1449 fixes it.

closed 2017-09-12 by jukkar

mike-scott, 2017-09-12:
Thanks @jukkar !

PR #1455 (2017-09-12): net: ethernet: Do not unref pending ARP request
If the packet is put pending because ARP request need to be
done, then do not unref original packet.

Fixes #1416
commit 0b8434f0 (2017-09-12): net: ethernet: Do not unref pending ARP request

If the packet is put pending because ARP request need to be
done, then do not unref original packet.

Fixes #1416