zephyrproject-rtos/zephyr #1418: kconfig options need some cleanup and reorganisation

nashif, 2017-09-08:
When you invoke 'make menuconfig' you get a view of the structure of the Kconfig variables in the kernel. Right now we have a few stray options that appear at the very top level when those should be hidden in sub-menus or should be depending on other features. Need to cleanup the structure and make sure we have all Kconfigs where they belong.

galak, 2017-09-12:
Any specific details on what options are stray?

nashif, 2017-09-12:
interrupt configuration, fatal handler... those are now fixed, still a few, see attachment.

aescolar, 2019-03-16:
@ulfalizer : Maybe you have a bit of time to check if this is consistent already?

nashif, 2021-05-28:
closing, lots of effort and work was put into this already..

closed 2021-05-28 by nashif