zephyrproject-rtos/zephyr #1420: LXR for Zephyr

psidhu, 2017-09-08:
I was advised to open an issue here for this.

It would be very helpful for developers, especially as Zephyr grows, to get [LXR](http://lxr.sourceforge.net/en/index.php) running and hosted on https://www.zephyrproject.org/.

Some examples:
- https://lxr.missinglinkelectronics.com/+trees
- http://elixir.free-electrons.com/linux/latest/source

My favorite is the missing link electronics one, but anything would be nice.

dbkinder, 2017-09-08:
Looks like LXR is a system that can create an interactive full cross-reference interface for all the Zephyr source code as an aid to viewing and understanding the source.  Perhaps a discussion from other developers would be good to help to identify the value of such an interface, and help prioritize this above, say, writing better overview documentation for Zephyr or improving the doxygen comments in the code.  I notice LXR is GPLv3 licensed, which may be an issue in our Apache2 license world.

andrewboie, 2017-09-08:
LXR license is irrelavent, its not like the code for it goes on the device.

> identify the value of such an interface, and help prioritize this above, say, writing better overview documentation for Zephyr

Is there something special about this bug that ties it into writing documentation? I'm confused.

psidhu, 2017-09-08:
@dbkinder if I want to develop Zephyr, I don't care much about 'overview documentation'. Instead, I want an aid that helps me during actual development of Zephyr, which is why something like LXR is useful and important (over grepping etc). It's less about "viewing and understanding the source".

For example, many Android developers use http://androidxref.com/ for the same reason. Linux developers use something like one of the examples I stated in the original posting.

Anyways, I hope that helps clear up any confusion.

dbkinder, 2017-09-08:
Anas tagged it as a "documentation" issue.  If there's some consensus that having LXR-generated documentation for Zephyr is a good thing, we can add it to our TODO list and get the devops team involved.  I'm just curious about the usefulness of this since it's the first time I've heard it come up.  I can see having a full cross-reference with a nice interface could be good for Zephyr developers (less so for Zephyr app developers I suspect.) Thanks for bringing it up for discussion.

dbkinder, 2017-09-09:
Have any of you explored alternative cross-reference tools, such as https://woboq.com/codebrowser.html ?  Woboq's code browser seems to not require the server-side resources (database and such) and seems more "modern" it appearance and behavior.

psidhu, 2017-09-11:
I was trying out woboq's example [here](https://code.woboq.org/qt5/qtbase/src/) and found that I don't think it's as fully-featured. Just going through it, it seems that its parser doesn't parse within `#DEFINE` blocks, meaning anything placed in those aren't searchable. I also found it's a bit cumbersome in that it requires extra clicks to do the same type of keyword search as in LXR.

The only other one I've looked into was OpenGrok, but it has a similar issue to woboq in that it doesn't list out all search terms found and doesn't persist that list across pages requiring you have to go back in order to research for the next found term.

dbkinder, 2017-09-12:
Looks like we'll need the devops team to evaluate this as LXR requires a database on the server and server configuration beyond my authority.  @inakypg @nashif

psidhu, 2017-12-05:
I'd like to bring this back to life a little bit. I noticed that the zephyrproject website has new UI changes, so I was wondering if this was hopefully next?

pfalcon, 2018-01-20:
Until we get something better, there's at least something: https://sourcegraph.com/github.com/zephyrproject-rtos/zephyr

The main issue with the service above is that there's no *cross-reference*, i.e. you can just from a call of function to its definition, etc.

However, there's a fast incremental search and inline "git blame" annotations, so again, for the lack of something better, it can be useful too, and available right away.

psidhu, 2018-01-22:
@dbkinder Hi, just wondering why the todo tag was removed? I still think it's something to be done.

@pfalcon Thanks for that, I never knew of the service. While it isn't a cross referencer, it's still helpful.

dbkinder, 2018-01-22:
"to do" means there's someone actively working on this (and there isn't to my knowledge)

psidhu, 2018-01-22:
Ah, thank you for the clarification. However, I was under the impression @inakypg and @nashif were supposed to be looking into this. Since I hadn't heard anything back from them, I figured it was still a WIP. 

That being said, what's the status of this if I haven't heard back from anyone?

henrikbrixandersen, 2018-02-25:
I have created an Elixir instance for zephyr on http://elixir.brixandersen.dk as proof-of-concept.

As can seen, the multiple tags (and multiple naming schemes for tags) causes two different "branches" to occur on the Elixir navigation tree. Not sure how to best handle them.

henrikbrixandersen, 2018-02-25:
@galak Can you shed some light on the significance of the `zephyr-vx.y.z` tags compared to the `vx.y.z` tags?

I am tempted to filter out the `zephyr-vx.y.z` tags in the Elixir configuration (as they seem redundant) and just display the `vx.y.z` tags.

henrikbrixandersen, 2018-02-25:
@galak Thank you for the explanation. So, from an Elixir point of view, filtering out the `zephyr-vx.y.z` sounds like a possible solution.

nashif, 2018-02-25:
@henrikbrixandersen yes

henrikbrixandersen, 2018-02-25:
I have opened a PR on free-electrons/elixir#35 for adding official support for the Zephyr Project in Elixir.

I will leave my proof-of-concept Elixir instance running on http://elixir.brixandersen.dk for now, but I would love to see it hosted elsewhere.

PR #6739 (2018-03-21): doc: Add Elixir cross-referencer to resources list
Bootlin is kindly providing an instance of their Elixir source code
cross-referencer tool with an index of the Zephyr kernel and samples
code.

This fixes #1420
commit abc47a07 (2018-03-21): doc: Add Elixir cross-referencer to resources list

Bootlin is kindly providing an instance of their Elixir source code
cross-referencer tool with an index of the Zephyr kernel and samples
code.

This fixes #1420
closed 2018-03-21 by nashif