zephyrproject-rtos/zephyr #1427: net: SYN/ACK packet sent by over SLIP has invalid Ethernet frame check

pfalcon, 2017-09-09:
While testing #1412, I paid attention that SYN/ACK (and only that) packet from Zephyr/qemu_x86 is painted black (the color of invalid checksum and other problems). Dump from Wireshark:

~~~
Frame 6: 82 bytes on wire (656 bits), 82 bytes captured (656 bits) on interface 0
Ethernet II, Src: USDepart_00:53:85 (00:00:5e:00:53:85), Dst: 62:84:a1:54:14:1b (62:84:a1:54:14:1b)
    Destination: 62:84:a1:54:14:1b (62:84:a1:54:14:1b)
    Source: USDepart_00:53:85 (00:00:5e:00:53:85)
    Type: IPv4 (0x0800)
    Padding: 000108000604
    Trailer: 000100005e005385c0000201000000000000
    Frame check sequence: 0xc0000202 [incorrect, should be 0x19ffca1b]
    [FCS Status: Bad]
~~~

Here's raw dump:

~~~
0000   62 84 a1 54 14 1b 00 00 5e 00 53 85 08 00 45 00  b..T....^.S...E.
0010   00 28 00 00 00 00 40 06 f6 cc c0 00 02 01 c0 00  .(....@.........
0020   02 02 1f 90 b3 d8 91 a3 e7 95 54 2d f0 6d 50 12  ..........T-.mP.
0030   05 00 95 91 00 00 00 01 08 00 06 04 00 01 00 00  ................
0040   5e 00 53 85 c0 00 02 01 00 00 00 00 00 00 c0 00  ^.S.............
0050   02 02                                            ..
~~~

Seen like that, it may turn out that we actually send out some garbage beyond the length of packet, which Wireshark decodes as padding/trailer/FCS.

Again, this isn't seen with other packet types (out of dozen total packets).

This is probably not top priority, but as we hopefully going to perfectalizing the stack phase, would be nice to figure out what happens here and clean it.

jukkar, 2017-09-12:
I could not reproduce this, tried with IPv4 and master commit d74f97d0699024fa8094e6f6703da38f63bc3f71
I wonder how this could be only be seen with SYN/ACK packet.

pfalcon, 2017-12-07:
Wasn't able to reproduce again.

closed 2017-12-07 by pfalcon