zephyrproject-rtos/zephyr #1429: LWM2M configs not defined

nashif, 2017-09-09:
LWM2M_BOOTSTRAP_SERVER not defined anywhere, yet being used here:

 subsys/net/lib/lwm2m/lwm2m_rd_client.c:#if defined(CONFIG_LWM2M_BOOTSTRAP_SERVER)

nashif, 2017-09-09:
same for 
LWM2M_SECURITY_OBJ_SUPPORT
LWM2M_DEVICE_PWRSRC_MAX

mike-scott, 2017-09-09:
Thanks @nashif.  I'll include fixing these in my next series.

nashif, 2017-10-30:
was this fixed?

mike-scott, 2017-10-30:
Not yet.  I can land a patch for this after the multi-fragment support is merged.

mike-scott, 2017-10-31:
These configs will be addressed in the upcoming patches to support LwM2M bootstrap.  If there's a priority I can remove them, but some of the code will come right back later.

mike-scott, 2018-02-08:
This adds the LWM2M_DEVICE_PWRSRC_MAX config:
https://github.com/zephyrproject-rtos/zephyr/pull/6042

nashif, 2018-11-05:
looks like this is resolved now

closed 2018-11-05 by nashif