zephyrproject-rtos/zephyr #1435: Could not connect to Eclipse Leshan Demo Server

roetzer, 2017-09-11:
Could not connect to Eclipse Leshan Demo Server

Serial Interface of Lwm2m Client:
[dev/eth_mcux] [DBG] eth_0_init: MAC e6:40:28:b3:b0:a1
[lib/lwm2m_engine] [DBG] lwm2m_engine_init: LWM2M engine thread started
[lwm2m_obj_security] [DBG] security_create: Create LWM2M security instance: 0
[lwm2m_obj_server] [DBG] server_create: Create LWM2M server instance: 0
[lwm2m_obj_device] [DBG] device_create: Create LWM2M device instance: 0
[lib/lwm2m_rd_client] [DBG] lwm2m_rd_client_init: LWM2M RD client thread started
[lwm2m_obj_firmware] [DBG] firmware_create: Create LWM2M firmware instance: 0
shell> [lwm2m-client] [INF] main: Run LWM2M client
[lib/lwm2m_engine] [DBG] lwm2m_engine_set: path:3/0/0, value:0x00014688, len:6
[lib/lwm2m_engine] [DBG] lwm2m_engine_set: path:3/0/1, value:0x00014695, len:23
[lib/lwm2m_engine] [DBG] lwm2m_engine_set: path:3/0/2, value:0x000146b3, len:9
[lib/lwm2m_engine] [DBG] lwm2m_engine_set: path:3/0/3, value:0x000146c3, len:3
[lib/lwm2m_engine] [DBG] lwm2m_engine_set: path:3/0/9, value:0x2001695f, len:1
[lib/lwm2m_engine] [DBG] lwm2m_engine_set: path:3/0/10, value:0x2001695c, len:4
[lib/lwm2m_engine] [DBG] lwm2m_engine_set: path:3/0/17, value:0x000146e6, len:16
[lib/lwm2m_engine] [DBG] lwm2m_engine_set: path:3/0/18, value:0x000146fe, len:5
[lib/lwm2m_engine] [DBG] lwm2m_engine_set: path:3/0/20, value:0x2001695f, len:1
[lib/lwm2m_engine] [DBG] lwm2m_engine_set: path:3/0/21, value:0x2001695c, len:4
[lib/lwm2m_engine] [DBG] lwm2m_engine_create_obj_inst: path:3303/0
[ipso_temp_sensor] [DBG] temp_sensor_create: Create IPSO Temperature Sensor instance: 0
[lib/lwm2m_engine] [DBG] lwm2m_engine_set: path:3303/0/5700, value:0x20016978, len:8
[lib/lwm2m_engine] [DBG] lwm2m_engine_create_obj_inst: path:3311/0
[ipso_light_control] [DBG] light_control_create: Create IPSO Light Control instance: 0
[lib/lwm2m_rd_client] [INF] lwm2m_rd_client_start: LWM2M Client: frdm_k64f-ipv4-3903044590
[lwm2m-client] [INF] main: IPv4 setup complete.
[lib/lwm2m_rd_client] [DBG] sm_do_init: RD Client started with endpoint 'frdm_k64f-ipv4-3903044590' and client lifetime 0
[dev/eth_mcux] [INF] eth_mcux_phy_event: Enabled 100M full-duplex mode.
[lib/lwm2m_rd_client] [DBG] sm_send_registration: registration sent [5.39.83.206]
[lib/lwm2m_engine] [DBG] lwm2m_udp_receive: udp_receive
[lib/lwm2m_engine] [DBG] lwm2m_udp_receive: checking for reply from [5.39.83.206]
[lib/lwm2m_engine] [ERR] lwm2m_udp_receive: No handler for response

--udp_request_handler: 5421; ZOAP_TYPE_CON: 3

mike-scott, 2017-09-11:
Hi, @roetzer

I'd like to confirm your local changes to the K64F .conf file.  Especially with regard to DHCP.  Without a gateway setting, the K64F will not connect to an external Leshan instance.  When I setup the example, I was testing Leshan locally on my laptop with private IPs on the same network.

mike-scott, 2017-09-11:
To confirm.  I was able to connect to the Leshan instance at 5.39.83.206 using the following config file changes:
https://hastebin.com/akajosahoy.diff

NOTE: I disabled IPv6 as there are 2 clients instantiated by the current sample and I wanted to make sure IPv4 connection was working.  This will go away with v1.10 and only 1 connection will be made via IPv4 or IPv6 (as determined automatically by the PEER IP setting).  The main changes were:
CONFIG_NET_DHCPV4=y
CONFIG_NET_APP_PEER_IPV4_ADDR="5.39.83.206"

mike-scott, 2017-09-18:
I believe this issue has been addressed and can be closed unless @roetzer has further comments.

roetzer, 2017-09-18:
Yes, your configuration has solved my problem. Thank you!

closed 2017-09-18 by mike-scott