zephyrproject-rtos/zephyr #1460: 1.10 Release Checklist

nashif, 2017-09-12:
- [ ] Release Notes
  - [ ] Update doc/release-notes.rst
  - [ ] Update doc/index.rst

dbkinder, 2017-09-12:
Concurrent editing loses your changes :(

galak, 2017-12-05:
@ruuddw, @nashif, @jhedberg, @jukkar, @carlescufi, @MaureenHelm, @andrewboie - Can you guys please take a look at the release notes for your section of ownership

carlescufi, 2017-12-05:
+1 on Bluetooth release notes (already ticked)

ruuddw, 2017-12-06:
+1 on release notes

galak, 2017-12-08:
@dbkinder I've tagged the release, can you handle the following tasks on the website:

 Update https://www.zephyrproject.org/developers/#releases-overview
 Update https://www.zephyrproject.org/developers/#downloads

galak, 2017-12-08:
@nashif can you update the doc generation for 1.10.0

nashif, 2017-12-08:
documentation updated

dbkinder, 2018-01-23:
Release done.

closed 2018-01-23 by dbkinder