zephyrproject-rtos/zephyr #1464: SYS_CLOCK_HW_CYCLES_PER_SEC is missing a default value

SebastianBoe, 2017-09-12:
Pretty much all Kconfig options have a default value, even when no reasonable default exists, e.g. 
ARCH defaults to x86, even though we have no reason to assume the user is using x86.

SYS_CLOCK_HW_CYCLES_PER_SEC is different, it has no default value.
Which is annoying if you use "make alldefconfig". Not having a default for this option
breaks "make alldefconfig", assuming alldefconfig is supported, and assuming the intention of alldefconfig is to give the user a default config without requiring user intervention.

Kbuild output when trying to use alldefconfig:
```
sebo@mach:~/workspace_zephyr/zephyr/samples/bluetooth/beacon$ rm -rf outdir/
sebo@mach:~/workspace_zephyr/zephyr/samples/bluetooth/beacon$ make
Using /home/sebo/workspace_zephyr/zephyr/boards/arm/nrf51_pca10028/nrf51_pca10028_defconfig as base
Merging prj.conf
#
# configuration written to .config
#
make[1]: Entering directory '/home/sebo/workspace_zephyr/zephyr'
make[2]: Entering directory '/home/sebo/workspace_zephyr/zephyr/samples/bluetooth/beacon/outdir/nrf51_pca10028'
  GEN     ./Makefile
scripts/kconfig/conf --silentoldconfig Kconfig
  DTC     dts/arm/nrf51_pca10028.dts_compiled
  CHK     include/generated/generated_dts_board.conf
  UPD     include/generated/generated_dts_board.conf
  CHK     include/generated/generated_dts_board.conf
  Using /home/sebo/workspace_zephyr/zephyr as source for kernel
  GEN     ./Makefile
  CHK     include/generated/version.h
  UPD     include/generated/version.h
  CHK     include/generated/generated_dts_board.h
  UPD     include/generated/generated_dts_board.h
  CHK     misc/generated/configs.c
  UPD     misc/generated/configs.c
  CHK     include/generated/offsets.h
  UPD     include/generated/offsets.h
  CC      lib/libc/minimal/source/stdlib/strtol.o
  CC      lib/libc/minimal/source/string/string.o
  CC      lib/libc/minimal/source/string/strncasecmp.o
  CC      lib/libc/minimal/source/stdout/sprintf.o
  CC      lib/libc/minimal/source/stdlib/strtoul.o
  CC      lib/libc/minimal/source/stdout/fprintf.o
  CC      lib/libc/minimal/source/stdout/prf.o
  CC      lib/libc/minimal/source/stdout/stdout_console.o
  CC      lib/libc/minimal/source/stdlib/atoi.o
  CC      lib/libc/minimal/source/string/strstr.o
  LD      lib/libc/minimal/source/stdlib/built-in.o
  LD      lib/libc/minimal/source/string/built-in.o
  LD      lib/libc/minimal/source/stdout/built-in.o
  LD      lib/libc/minimal/source/built-in.o
  LD      lib/libc/minimal/built-in.o
  LD      lib/libc/built-in.o
  LD      lib/built-in.o
  CC      kernel/version.o
  CC      kernel/alert.o
  CC      kernel/atomic_c.o
  CC      kernel/device.o
  CC      kernel/idle.o
  CC      kernel/errno.o
  CC      kernel/init.o
  CC      kernel/mailbox.o
  CC      kernel/mem_slab.o
  CC      kernel/mempool.o
  CC      kernel/msg_q.o
  CC      kernel/mutex.o
  CC      kernel/pipes.o
  CC      kernel/poll.o
  CC      kernel/queue.o
  CC      kernel/sched.o
  CC      kernel/sem.o
  CC      kernel/stack.o
  CC      kernel/sys_clock.o
[... 169 lines cut ...]
  UPD     include/generated/generated_dts_board.h
  CHK     misc/generated/configs.c
  UPD     misc/generated/configs.c
In file included from /home/sebo/workspace_zephyr/zephyr/include/arch/x86/irq_controller.h:23:0,
                 from /home/sebo/workspace_zephyr/zephyr/include/arch/x86/arch.h:18,
                 from /home/sebo/workspace_zephyr/zephyr/include/arch/cpu.h:13,
                 from /home/sebo/workspace_zephyr/zephyr/include/kernel.h:3805,
                 from /home/sebo/workspace_zephyr/zephyr/kernel/include/kernel_structs.h:10,
                 from /home/sebo/workspace_zephyr/zephyr/arch/x86/core/offsets/offsets.c:29:
/home/sebo/workspace_zephyr/zephyr/include/drivers/sysapic.h: In function ‘__irq_controller_eoi’:
/home/sebo/workspace_zephyr/zephyr/include/drivers/sysapic.h:35:20: error: ‘CONFIG_LOAPIC_BASE_ADDRESS’ undeclared (first use in this function)
  *(volatile int *)(CONFIG_LOAPIC_BASE_ADDRESS + LOAPIC_EOI) = 0;
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/sebo/workspace_zephyr/zephyr/include/drivers/sysapic.h:35:20: note: each undeclared identifier is reported only once for each function it appears in
/home/sebo/workspace_zephyr/zephyr/./Kbuild:35: recipe for target 'arch/x86/core/offsets/offsets.o' failed
make[3]: *** [arch/x86/core/offsets/offsets.o] Error 1
/home/sebo/workspace_zephyr/zephyr/Makefile:1085: recipe for target 'prepare' failed
make[2]: *** [prepare] Error 2
make[2]: Leaving directory '/home/sebo/workspace_zephyr/zephyr/samples/bluetooth/beacon/outdir/nrf51_pca10028'
Makefile:178: recipe for target 'sub-make' failed
make[1]: *** [sub-make] Error 2
make[1]: Leaving directory '/home/sebo/workspace_zephyr/zephyr'
/home/sebo/workspace_zephyr/zephyr/Makefile.inc:82: recipe for target 'all' failed
make: *** [all] Error 2
```

dbkinder, 2017-09-12:
FYI, There are lots of Kconfig options defined (681) without default values in the same block.  In kernel/Kconfig alone there are these:

     no default for BUILD_TIMESTAMP in kernel/Kconfig
     no default for ATOMIC_OPERATIONS_BUILTIN in kernel/Kconfig
     no default for ATOMIC_OPERATIONS_CUSTOM in kernel/Kconfig  
     no default for ATOMIC_OPERATIONS_C in kernel/Kconfig
     no default for MEM_POOL_SPLIT_BEFORE_DEFRAG in kernel/Kconfig
     no default for MEM_POOL_DEFRAG_BEFORE_SPLIT in kernel/Kconfig     
     no default for MEM_POOL_SPLIT_ONLY in kernel/Kconfig
     no default for SYS_CLOCK_HW_CYCLES_PER_SEC in kernel/Kconfig
     no default for XIP in kernel/Kconfig

Should all of these need default values defined too?

nashif, 2017-09-12:
> Should all of these need default values defined too?

No, they should not. Booleans are different, the issue here is about an option that accepts a value.

dbkinder, 2017-09-13:
Ah.  OK.  In that case only 129 config values without a default (ignoring bool values) and only ``SYS_CLOCK_HW_CYCLES_PER_SEC`` in kernel/Kconfig

Let me know if having the list of values without defaults would help anyone.

SebastianBoe, 2018-09-06:
This issue was provoked by alldefconfig and alldefconfig has been dropped. The underlying issue still exists, but this issue does not demonstrate (or reproduce) it well, so it is cleanest to close this.

closed 2018-09-06 by SebastianBoe

neuberfran, 2023-10-25:
I'm having this issue in 2023 (I absolutely need) Zephyr 1.13 and not the current one. The issue appears when I try to change my device from warp7_m4 (it works here) to imx7d-pico. Here the make command does not work:

https://gist.github.com/neuberfran/62b7856b465c06ee65f7a97491e179b5

[image]

neuberfran, 2023-11-04:
@SebastianBoe How to solve it? I stay using zephyr 1.13 and I have this issue in 2023. I need it:
[image]

neuberfran, 2023-11-23:
Important updates to my issue: When I enable the lines below again, the mcu is no longer recognized (as it was being wrongly) as x86. And the error in the photos below appears in the command cmake -DBOARD=pico_pi_m4 ../..
before the command:
make

> if !XIP
> config FLASH_SIZE
> 	default 0
> config FLASH_BASE_ADDRESS
> 	default 0
> endif

Therefore, the correction of my issue involves a correct reset of the pico_pi_m4(imx7d-pico) board in Zephyr 1.13 so that the rpmsg_lite app can function and generate correctly.

> git clone https://github.com/neuberfran/zephyr_old_version
> 
> source /home/neuberfran/diegosueiro3/zephyr/zephyr-env.sh
> export GCCARMEMB_TOOLCHAIN_PATH=/home/neuberfran/Downloads/gcc-arm-none-eabi-10.3-2021.10
> 
> export ZEPHYR_TOOLCHAIN_VARIANT=gccarmemb
> 
> mkdir -p  build/pico_pi_m4 && cd build/pico_pi_m4
> mkdir -p  build/colibri_imx7d_m4 && cd build/colibri_imx7d_m4
> mkdir -p build/warp7_m4 && cd build/warp7_m4
> 
> cmake -DBOARD=pico_pi_m4 ../..
> cmake -DBOARD=colibri_imx7d_m4 ../..
> cmake -DBOARD=warp7_m4 ../..
> 
> make

[image]
[image]

neuberfran, 2023-11-23:
When I try **west build -p auto -b warp7_m4 remote_echo**  in Zephry 3.3 When I run **west build -p auto -b pico_pi_m4 remote_echo** (Even without satisfactory results when I place the zephyr.bin resulting from this command in U-boot) **I don't get errors.**

> west build -p auto -b warp7_m4 remote_echo 
> -- west build: generating a build system
> Loading Zephyr default modules (Zephyr base (cached)).
> -- Application: /home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/remote_echo
> -- CMake version: 3.24.1
> -- Cache files will be written to: /home/neuberfran/.cache/zephyr
> -- Zephyr version: 3.3.99 (/home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/zephyr)
> -- Found west (found suitable version "1.0.0", minimum required is "0.7.1")
> -- Board: warp7_m4
> -- Found host-tools: zephyr 0.16.0 (/home/neuberfran/zephyr-sdk-0.16.0)
> -- Found toolchain: zephyr 0.16.0 (/home/neuberfran/zephyr-sdk-0.16.0)
> -- Found BOARD.dts: /home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/zephyr/boards/arm/warp7_m4/warp7_m4.dts
> devicetree error: /home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/zephyr/boards/arm/warp7_m4/warp7_m4.dts:92 (column 1): parse error: undefined node label 'mub'
> -- In: /home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/build/zephyr, command: /usr/bin/python3.8;/home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/zephyr/scripts/dts/gen_defines.py;--dts;/home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/build/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;/home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/zephyr/dts/bindings;--header-out;/home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/build/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/build/zephyr/zephyr.dts.new;--edt-pickle-out;/home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/build/zephyr/edt.pickle;--vendor-prefixes;/home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/zephyr/dts/bindings/vendor-prefixes.txt
> CMake Error at /home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/zephyr/cmake/modules/dts.cmake:261 (message):
>   gen_defines.py failed with return code: 1
> Call Stack (most recent call first):
>   /home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/zephyr/cmake/modules/zephyr_default.cmake:115 (include)
>   /home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
>   /home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
>   /home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/zephyr/cmake/app/boilerplate.cmake:25 (find_package)
>   CMakeLists.txt:4 (include)
> 
> 
> -- Configuring incomplete, errors occurred!
> FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/build -GNinja -S/home/neuberfran/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite/remote_echo
> neuberfran@neuberfran2004:~/diegosueiro/rpmsg-lite-zephyr-support/zephyr/samples/subsys/ipc/rpmsg_lite$

neuberfran, 2023-11-25:
Updates:

[image]