zephyrproject-rtos/zephyr #1488: Replacing Make/Kbuild with CMake

nashif, 2017-09-13:
This PR implements RFC: [Replacing Make/Kbuild with CMake](https://lists.zephyrproject.org/pipermail/zephyr-devel/2017-April/007538.html)

The Gerrit feedback has been addressed, but the work is not complete. It is uploaded in an incomplete form to enable collaboration and early feedback.

### Work In Progress
#### Blocker TODOs
TODOs that must be fixed before the PR is merged.

- [x] Support Device tree
- [x] Support generating isr_tables
- [x] Support generating offsets.c
- [x] Integrate with sanitycheck.
  - [ ] Verify that all samples are similar enough[0] before and after CMake
- [x] Integrate with qemu
  - [ ] Integrate with openocd
- [x] Integrate with Zephyr CI.
- [ ] Update docs.
- [ ] Document how to develop CMakeLists.txt code
- [ ] Remove KBuild
- [ ] Check if all Make CLI arguments are supported by an equivalent CMake feature
- [x] Support configuring the build for code size, speed, unoptimzed, etc.
- [x] Port kernel
- [ ] Port arch
  - [x] arm
  - [ ] arc
  - [ ] nios2
  - [x] riscv32
  - [x] x86
  - [ ] xtensa
- [ ] Port boards
  - [ ] arm
  - [ ] x86
  - [ ] xtensa
  - [x] riscv32
  - [ ] arc
  - [ ] nios2
- [ ] Port ext
- [ ] Port subsys
- [ ] Port samples
- [ ] Port tests
- [ ] Have CMake recursively call CMake to build host tools instead of
  requiring the user to first invoke a seperate CMake build system to
  build the host tools.
- [x] Support default BOARD's in application CMakeLists.txt code.[1]
- [x] Fix the CLI usability problem: -DCMAKE_TOOLCHAIN_FILE=$ZEPHYR_BASE/cmake/toolchain.cmake
- [ ] Test with Linux,OSX,Windows
- [ ] Test with Ninja,Make
- [ ] Test with all supported toolchains

#### Non-blocker TODOs
TODOs that must be fixed, but may be delayed until after the PR is merged.

- [ ] Check if it is acceptable to use an older version of
  CMake. Currently minumum required version is 3.7, but the latest
  Ubuntu LTS ships with 3.5. Supporting an older version would make it
  easier for users to get started with Zephyr.
- [ ] Ensure that verbose and normal builds are logging the appropriate amount of information.

#### Un-classified TODOs
TODOs in need of being classified as Blocker or non-blocker.

- [ ] Evaluate how application CMakeLists.txt code should integrate
  with Zephyr CMakeLists.txt code. Currently they interact in a manner
  very similair to how KBuild worked, but even though this worked for
  KBuild, it might not be the best solution for a CMake build system.
In particular, consider the "bootloader use-case" of a multi-program
build.
- [ ] Support KBuild's ld-option (The ability to test if the linker
  supports a flag).
- [x] Introduce more CMake extentions to reduce boilerplate and enforce
  conventions in driver/ subsys/ ext/ CMakeLists.txt code.
- [ ] Integrate more Kconfig targets than menuconfig

### Getting started
Install CMake (>=3.7)
Install Python 3 (>=3.4)

Optionally install Ninja, recommended for most users.
To enable Ninja add the flag -GNinja when calling cmake.

#### Windows
Open Windows Command Prompt:

> set ZEPHYR_BASE=<zephyr root>
> set ZEPHYR_GCC_VARIANT=gccarmemb
> set GCCARMEMB_TOOLCHAIN_PATH=<path to toolchain>
> set BOARD=nrf52_pca10040

> cd samples\bluetooth\beacon
> cmake -Bbuild -H.
> cmake --build build

#### Linux
$ source zephyr-env.sh

$ cd $ZEPHYR_BASE/scripts
$ mkdir build && cd build
$ cmake .. && make
$ export PREBUILT_HOST_TOOLS=$(pwd)

$ export BOARD=nrf52_pca10040
$ export ZEPHYR_SDK_INSTALL_DIR=<path to zephyr sdk>
$ export ZEPHYR_GCC_VARIANT=zephyr

$ cd $ZEPHYR_BASE/samples/bluetooth/beacon
$ cmake -Bbuild -H.
$ cmake --build build

[0] Binary identical FW's before and after the CMake transition would
make verification easy. But it may not be what we really want.
[1] https://github.com/zephyrproject-rtos/zephyr/pull/1107#issuecomment-322463484

oska874, 2017-09-18:
Why want to replace kconfig with cmake? I think kconfig is good and easy to use .

carlescufi, 2017-11-09:
Of the blocker todos, everything is done AFAICT except the documentation.

Carles: samples/**/*.rst (except samples/net)
@jukkar: samples/net/**/*.rst
@mbolivar: boards/**/*.rst

Anything else I'm missing?

Zarkopafilis, 2017-11-15:
Why Install CMake (>=3.7) ? Current apt version is way lower

carlescufi, 2017-11-15:
@Zarkopafilis I believe it's due to requiring add_custom_command() to use DEPFILE, which was introduced in CMake 3.7. The getting started guide gives you a step by step guide on installing CMake from cmake.org, and it's quite simple.

Zarkopafilis, 2017-11-15:
Ok. Sidenote: On windows with MSYS it works with cmake 3.6.2.... on a
docker container with an ubuntu base image it did not.

carlescufi, 2017-11-15:
@Zarkopafilis that's because the CMake scripts detect that you're running MSYS and lower the required version. It's an exception in the case of MSYS, which doesn't support Ninja. This is temporary since we will soon switch to native CMake with Ninja and no MSYS at all on Windows.

closed 2017-11-20 by SebastianBoe