zephyrproject-rtos/zephyr #1495: esp32: newlibc errors

mrrosen, 2017-09-13:
When trying to use newlibc with ESP32, its possible to get linker errors when using some libc functions, such as printf. There are a number of missing function implementations for __getreent and _sbrk_r. It appears that the libc library used in the esp-idf toolchain was compiled with different options than the ones used in the zephyr sdk; namely that the esp-idf libc seems to use the REENTRANT_SYSCALLS_PROVIDED option for its newlibc while zephyr's sdk does not. It seems Zephyr for ESP32 doesnt fully support newlibc yet; are there plans to fully enable this in the future?

lpereira, 2017-09-15:
Yes, there are plans, since newlib must be supported by all boards supported by Zephyr.  However, I'm currently enabling the hardware, so this might take a while to happen.

pille5, 2018-09-01:
I've just ran into the same issue. `Zephyr.js` does not build for ESP32. Is there any workaround?

nashif, 2019-01-30:
currently not supported.

closed 2019-01-30 by nashif

GuptaAshi-Eaton, 2022-04-25:
Is newlibc library supported for ESP32 in Zephyr ?

xyzzy42, 2023-05-17:
It's been a year.  Is it supported?  I'm not sure, but maybe.  Here's what I've found.

I built my own ESP32 toolchain using [crosstool-ng](https://github.com/crosstool-ng/crosstool-ng).  This includes newlib.
I used this with Zephyr via `ZEPHYR_TOOLCHAIN_VARIANT=cross-compile`.
This works initially only with the minimal libc option.  Newlib is not selectable as zephyr libc choice.  This appears to be the expected behavior for "cross-compile" toolchains.
Passing the option `-DTOOLCHAIN_HAS_NEWLIB=on` to cmake will allow newlib to be selected as the libc.
However, Zephyr fails to link, with this same issue as six years ago.

I then added trivial implementations of `pthread_setcancelstate()` and `__getreent()` to my app, using those from the ESP-IDF's bootloader example.

And it works.

However, I'm not using the ESP32 in SMP mode.  I don't think ESP32 SMP works in Zephyr yet.  I doubt that the trivial implementation of `__getreent()` would not produce working reentrant libc functions in SMP mode.  In fact, I suspect it doesn't even work with multiple threads in non-SMP mode.

From what I can see, the zephyr sdk for esp32 does use newlib.  But it's an older major version than in crosstool-ng.  It looks to me that it avoids the __getreent issue by using a version provided by newlib, rather than expecting something like Zephyr or ESP-IDF to provide it.  I'm not convinced this actually works either!