zephyrproject-rtos/zephyr #1511: qemu_nios2 should use the GHRD design

andrewboie, 2017-09-14:
We currently use a custom Nios II build and have had to patch QEMU in order to support the machine type.
We would rather use vanilla QEMU with the GHRD:
Add a SOC target for the GHRD, see if we can find a system.h and layout.h for it so we can run Zephyr in QEMU with the default GHRD machine.

ramakrishnapallala, 2017-09-19:
To use default qemu GHRD config, we should also need to use the ghrd softcpu image. Because changing the system.h and layout.h may not be sufficient enough to use the default qemu ghrd target.

I will do little more research and update the issue accordingly.

ramakrishnapallala, 2017-10-16:
Hi Anas,

I see that you have released the new SDK 0.92 excluding the Qemu v2.10 which it will have the older version only.

Now coming back to this issue, should try to address this issue by next SDK release and include the qemu v2.10?  Right now I am trying to make nios2 qemu work with upstream one by making the base address chage for the relevant IP's.

ramakrishnapallala, 2017-10-23:
Hi Anas, I think we should not this issue to Zephyr release, instead we should link it to SDK relea

ramakrishnapallala, 2019-01-10:
Not working on Nios-II MAX10 at present.

closed 2019-12-18 by nashif