zephyrproject-rtos/zephyr #1522: "make qemu" may not regenate .config after changes to prj.conf

pfalcon, 2017-09-15:
I noticed that "make qemu", which was deprecated, but is still there, may not propagate changes if prj.conf was updated (i.e. it would not cause .config to be updated and/or an app to be rebuilt). Let's either make it behave the same as "make run", or remove completely.

nashif, 2017-09-15:
make qemu does exist anymore, what makes you say it is still available?

pfalcon, 2017-09-15:
The fact that I very well remember when Zephyr had it, and so still occasionally use it, and it "works" (runs QEMU as expected). So, let's either kill it completely (so for example "make qemu" printed "Error: use 'make run' instead"), or let's maintain it in a deprecated shape, so at least it didn't do harm to folks for whom it's being maintained.

nashif, 2017-09-16:
'make qemu' does not work anymore, it was removed after being deprecated for a while. You should be using 'make run' now.

pfalcon, 2017-09-17:
> 'make qemu' does not work anymore

It does for me:

~~~
$ cd samples/net/echo_server
#  don't trust Zephyr's *clean
$ rm -rf outdir
$ make
$ make qemu
make[1]: Entering directory '/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr'
make[2]: Entering directory '/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr/samples/net/echo_server/outdir/qemu_x86'
  CHK     include/generated/generated_dts_board.conf
  Using /mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr as source for kernel
  GEN     ./Makefile
  CHK     include/generated/version.h
  CHK     include/generated/generated_dts_board.h
  CHK     misc/generated/configs.c
  CHK     include/generated/offsets.h
To exit from QEMU enter: 'CTRL+a, x'
[QEMU] CPU: qemu32
qemu-system-i386: warning: Unknown firmware file in legacy mode: genroms/multiboot.bin

shell> [net/app] [INF] net_app_init: Initializing network
[net/app] [INF] setup_ipv4: IPv4 address: 192.0.2.1
[net/app] [INF] ipv6_event_handler: IPv6 address: fe80::200:5eff:fe00:53e5
[net/app] [INF] ipv6_event_handler: IPv6 address: fe80::200:5eff:fe00:53e5
~~~

So, if someone has an older script which uses "make qemu" or reads older instructions using it, they may be affected by an issue in the subj.

nashif, 2017-09-18:
ok, we forgot a target in 1 spot, scripts/Makefile.qemu

closed 2017-09-19 by nashif