zephyrproject-rtos/zephyr #1526: Bluetooth:mesh:prov_start: Invalid authentication

wayen30, 2017-09-16:
I test sample/bluetooth/mesh with a ble-board and an android phone(bluetooth mesh apk from silicon lab ). 

----------------------------------------------------------------------------------
phone: Provisioning invite
----------------------------------------------------------------------------------
ble board: Provisioning capabilities: 01 01 0001 00 00 04 0008 00 0000
----------------------------------------------------------------------------------
phone: Provisioning start:
[bt] [DBG] prov_start: (0x20031114) Algorithm:   0x00
[bt] [DBG] prov_start: (0x20031114) Public Key:  0x00
[bt] [DBG] prov_start: (0x20031114) Auth Method: 0x02
[bt] [DBG] prov_start: (0x20031114) Auth Action: 0x00
[bt] [DBG] prov_start: (0x20031114) Auth Size:   0x04
----------------------------------------------------------------------------------
ble board: prov_start: Invalid authentication method: 0x02; action: 0x00; size: 0x04
----------------------------------------------------------------------------------

Is there a problem with the parameters sent from apk for provisioning start?

Laczen, 2017-09-16:
Hi,

The app from silicon labs cannot handle the authentication method proposed by the mesh node. To get it working you need to change the settings in the provisioning structure:
.output_size = 0,
//.output_actions= BT_MESH_DISPLAY_NUMBER,
//.output_number = output_number,
Then it should work.

Please take into account that this provisioning method is insecure, and you should not use it in production.

wayen30, 2017-09-18:
@Laczen thanks!
Is there any other bluetooth mesh apk?

nvl1109, 2017-11-27:
Another way to get it works:
```
	.output_size = 4,
	.output_actions = BT_MESH_DISPLAY_NUMBER | BT_MESH_BEEP | BT_MESH_VIBRATE | BT_MESH_BLINK,
	.output_number = output_number,
```
You need to enter the OOB number (get from zephyr console) when Silicon Lab app requires.

shukla-ashish-92, 2017-11-29:
.output_size = 4,
.output_actions = BT_MESH_DISPLAY_NUMBER | BT_MESH_BEEP | BT_MESH_VIBRATE | BT_MESH_BLINK, 
.output_number = output_number,

by making these changes in zephyr/samples/bluetooth/mesh/src/main.c, silicon app asks for OOB number, but it fails to configure the node.

jhedberg, 2018-02-08:
The SiLabs application's bug is a known one. I don't directly see any issue we can fix on the Zephyr side here. The provisioning protocol has been successfully run against all the qualification (PTS) test cases, so from the perspective the implementation should be correct.

nashif, 2018-02-08:
then this bug should be closes as far as I a concerned.

jhedberg, 2018-02-13:
Closing, since this seems to be a known issue on the SiLabs app side

closed 2018-02-13 by jhedberg