zephyrproject-rtos/zephyr #1527: make htmldocs failed

oska874, 2017-09-18:
I want to make html docs of zephyr 1.9 , follow the procedure of https://www.zephyrproject.org/doc/README.html?highlight=htmldocs ,but failed finally. The following text is the error log printed in shell .

My environment is Ubuntu 14 (tried 16.04 but fail too,error log almost the same), zephyr code is the newest.

```
zephyr$ make htmldocs
/usr/local/lib/python2.7/dist-packages/sphinx/util/requests.py:62: UserWarning: Some links may return broken results due to being unable to check the Server Name Indication (SNI) in the returned SSL cert against the hostname in the url requested. Recommended to install requests-2.4.1+.
  'Some links may return broken results due to being unable to '

/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:27: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:27: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:27: WARNING: Invalid definition: Expected end of definition. [error at 12]
  bt_conn_info.__unnamed__
  ------------^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:33: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:33: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:33: WARNING: Invalid definition: Expected end of definition. [error at 23]
  bt_gatt_discover_params.__unnamed__
  -----------------------^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:33: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:33: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:33: WARNING: Invalid definition: Expected end of definition. [error at 19]
  bt_gatt_read_params.__unnamed__
  -------------------^
/os_dev/yoc2/home/root/zephyr/doc/api/io_interfaces.rst:72: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/io_interfaces.rst:72: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/io_interfaces.rst:72: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/io_interfaces.rst:72: WARNING: Invalid definition: Expected end of definition. [error at 18]
  uart_device_config.__unnamed__
  ------------------^
/os_dev/yoc2/home/root/zephyr/doc/api/networking.rst:25: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/networking.rst:25: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/networking.rst:25: WARNING: Invalid definition: Expected end of definition. [error at 7]
  net_buf.__unnamed__
  -------^
/os_dev/yoc2/home/root/zephyr/doc/api/networking.rst:37: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/networking.rst:37: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/networking.rst:37: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/networking.rst:37: WARNING: Invalid definition: Expected end of definition. [error at 8]
  in6_addr.in6_u
  --------^
/os_dev/yoc2/home/root/zephyr/doc/api/networking.rst:37: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]
[... 803 lines cut ...]
          ---------------------------------^
      If declId, parameters, and qualifiers:
        Invalid definition: Expected identifier in nested name. [error at 33]
          NET_MGMT_DEFINE_REQUEST_HANDLER((( NET_MGMT_IFACE_BIT | NET_MGMT_LAYER ( NET_MGMT_LAYER_L2 )| NET_MGMT_LAYER_CODE (0x154))|NET_REQUEST_IEEE802154_CMD_SET_TX_POWER))
          ---------------------------------^
      If parenthesis in noptr-declarator:
        Error in declarator or parameters and qualifiers
        If pointer to member declarator:
          Invalid definition: Expected '::' in pointer to member (function). [error at 54]
            NET_MGMT_DEFINE_REQUEST_HANDLER((( NET_MGMT_IFACE_BIT | NET_MGMT_LAYER ( NET_MGMT_LAYER_L2 )| NET_MGMT_LAYER_CODE (0x154))|NET_REQUEST_IEEE802154_CMD_SET_TX_POWER))
            ------------------------------------------------------^
        If declarator-id:
          Invalid definition: Expecting "(" in parameters_and_qualifiers. [error at 54]
            NET_MGMT_DEFINE_REQUEST_HANDLER((( NET_MGMT_IFACE_BIT | NET_MGMT_LAYER ( NET_MGMT_LAYER_L2 )| NET_MGMT_LAYER_CODE (0x154))|NET_REQUEST_IEEE802154_CMD_SET_TX_POWER))
            ------------------------------------------------------^
/os_dev/yoc2/home/root/zephyr/doc/api/networking.rst:112: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/networking.rst:112: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/networking.rst:112: WARNING: Invalid definition: Expected end of definition. [error at 21]
  ieee802154_req_params.__unnamed__
  ---------------------^
```

oska874, 2017-09-18:
Then I tried work with `virtualenv` ,re-installed tool with command "pip install -r ./scripts/requirements.txt" , but failed again.

```
 make htmldocs
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:27: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:27: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:27: WARNING: Invalid definition: Expected end of definition. [error at 12]
  bt_conn_info.__unnamed__
  ------------^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:33: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:33: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:33: WARNING: Invalid definition: Expected end of definition. [error at 23]
  bt_gatt_discover_params.__unnamed__
  -----------------------^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:33: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:33: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
/os_dev/yoc2/home/root/zephyr/doc/api/bluetooth.rst:33: WARNING: Invalid definition: Expected end of definition. [error at 19]
  bt_gatt_read_params.__unnamed__
  -------------------^
/os_dev/yoc2/home/root/zephyr/doc/api/io_interfaces.rst:60: WARNING: Error in type declaration.
If typedef-like declaration:
  Type must be either just a name or a typedef-like declaration.
  If just a name:
    Error in declarator or parameters and qualifiers
    Invalid definition: Expected identifier in nested name, got keyword: int [error at 3]
      int(* spi_api_io_async)(struct spi_config *config, const struct spi_buf *tx_bufs, size_t tx_count, struct spi_buf *rx_bufs, size_t rx_count, struct k_poll_signal *async)
      ---^
  If typedef-like declaration:
    Error in declarator
    If pointer to member declarator:
      Invalid definition: Expected identifier in nested name. [error at 3]
        int(* spi_api_io_async)(struct spi_config *config, const struct spi_buf *tx_bufs, size_t tx_count, struct spi_buf *rx_bufs, size_t rx_count, struct k_poll_signal *async)
        ---^
    If declId, parameters, and qualifiers:
      Invalid definition: Expected identifier in nested name. [error at 3]
        int(* spi_api_io_async)(struct spi_config *config, const struct spi_buf *tx_bufs, size_t tx_count, struct spi_buf *rx_bufs, size_t rx_count, struct k_poll_signal *async)
        ---^
    If parenthesis in noptr-declarator:
      Error in declarator
      If pointer to member declarator:
        Invalid definition: Expected identifier in nested name. [error at 23]
          int(* spi_api_io_async)(struct spi_config *config, const struct spi_buf *tx_bufs, size_t tx_count, struct spi_buf *rx_bufs, size_t rx_count, struct k_poll_signal *async)
          -----------------------^
      If declId, parameters, and qualifiers:
        Invalid definition: Expecting "," or ")" in parameters_and_qualifiers, got "c". [error at 43]
          int(* spi_api_io_async)(struct spi_config *config, const struct spi_buf *tx_bufs, size_t tx_count, struct spi_buf *rx_bufs, size_t rx_count, struct k_poll_signal *async)
          -------------------------------------------^
      If parenthesis in noptr-declarator:
        Invalid definition: Expected ')' in "( ptr-declarator )" [error at 24]
          int(* spi_api_io_async)(struct spi_config *config, const struct spi_buf *tx_bufs, size_t tx_count, struct spi_buf *rx_bufs, size_t rx_count, struct k_poll_signal *async)
          ------------------------^
If type alias or template alias:
  Invalid definition: Expected identifier in nested name, got keyword: int [error at 3]
    int(* spi_api_io_async)(struct spi_config *config, const struct spi_buf *tx_bufs, size_t tx_count, struct spi_buf *rx_bufs, size_t rx_count, struct k_poll_signal *async)
    ---^
/os_dev/yoc2/home/root/zephyr/doc/api/io_interfaces.rst:72: WARNING: Invalid definition: Expected identifier in nested name. [error at 0]

  ^
[... 943 lines cut ...]
        ---^
    If declId, parameters, and qualifiers:
      Invalid definition: Expected identifier in nested name. [error at 3]
        int(* http_send_data_t)(struct net_pkt *pkt, net_context_send_cb_t cb, s32_t timeout, void *token, void *user_data)
        ---^
    If parenthesis in noptr-declarator:
      Error in declarator
      If pointer to member declarator:
        Invalid definition: Expected identifier in nested name. [error at 23]
          int(* http_send_data_t)(struct net_pkt *pkt, net_context_send_cb_t cb, s32_t timeout, void *token, void *user_data)
          -----------------------^
      If declId, parameters, and qualifiers:
        Invalid definition: Expecting "," or ")" in parameters_and_qualifiers, got "p". [error at 40]
          int(* http_send_data_t)(struct net_pkt *pkt, net_context_send_cb_t cb, s32_t timeout, void *token, void *user_data)
          ----------------------------------------^
      If parenthesis in noptr-declarator:
        Invalid definition: Expected ')' in "( ptr-declarator )" [error at 24]
          int(* http_send_data_t)(struct net_pkt *pkt, net_context_send_cb_t cb, s32_t timeout, void *token, void *user_data)
          ------------------------^
If type alias or template alias:
  Invalid definition: Expected identifier in nested name, got keyword: int [error at 3]
    int(* http_send_data_t)(struct net_pkt *pkt, net_context_send_cb_t cb, s32_t timeout, void *token, void *user_data)
    ---^

```

pfalcon, 2017-09-18:
Are you sure it's "failed"? The warning/errors above look "normal" (i.e. they're expected), can't tell if there's one which really causes failure.

dbkinder, 2017-09-18:
The doc build actually did succeed, but all the warnings are unnerving.  I've submitted a patch to the Makefile to run the filter-known-issues script we use during CI processing to hide "expected" warnings.

oska874, 2017-09-19:
So the document made successfully actually.I calcuate the size of zephyr `html`, it's 100M.
But when I open the index.html, it's almost empty, like this:

[image]

Is this correct?

dbkinder, 2017-09-19:
Nope.  You found the doxygen-generated API document tree, which we really don't use.  The generated documents would be in doc/_build/html/index.html

oska874, 2017-09-19:
Yes，I get from there. And this is what it does.

dbkinder, 2017-09-19:
To be clear, after the local doc generation (make htmldocs) is done, the Zephyr document tree is in doc/_build/html/index.html.  (This document set uses a simplified generic theme.)   There is also a set of doxygen API files built in doc/doxygen/html/index.html (mostly for debugging use), but this doxygen API html set is not published to the website.  Once PR #1541 is merged, the "expected" warnings will be filtered from the "make htmldocs" output.

PR #1592 (2017-09-22): doc: include message filtering in doc build script
Because of known issues with Sphinx/Breathe tools we're using to
generate doxygen-based comments for our API documentation, we're getting
a bevy of warning messages written out. As a workaround for our CI
system, we created a filter-known-issues.py script to remove "expected"
warnings from the output.

This patch moves calling that filter script into the doc generation
Makefile so folks making local builds of the docs won't be tripped up by
all the warning messages either.  Output of the "make htmldocs" command
is now filtered so only "unexpec [... cut]
commit d3b0cd48 (2017-09-22): doc: move docgen filter out of shippable script

Because of known issues with Sphinx/Breathe tools we're using to
generate doxygen-based comments for our API documentation, we're getting
a bevy of warning messages written out. As a workaround for our CI
system, we created a filter-known-issues.py script to remove "expected"
warnings from the output.

This patch moves calling that filter script into the doc generation
Makefile so folks making local builds of the docs won't be tripped up by
all the warning messages either.  Output of the "make htmldocs" command
is now filtered so only "unexpecte [... cut]
closed 2017-09-22 by nashif