zephyrproject-rtos/zephyr #1533: It is not documented how to discover, install, and use external toolchains

SebastianBoe, 2017-09-18:
Currently each board will describe how to use an external toolchain.

This causes duplication between boards, as they re-iterate what toolchains are available and how to use them, and it causes some users to not discover that there are more supported toolchains available than what the board's documentation mentions.

pfalcon, 2017-09-18:
Are there some examples? In my list, Zephyr SDK is the canonical source of the toolchain, and I don't remember seeing my of external toolchain docs in the boards (I may imagine some, like esp32, indeed have that).

SebastianBoe, 2017-09-18:
The board xtensa/esp32 is describing how to install the ZEPHYR_GCC_VARIANT=espressif toolchain.

The board riscv32/zedboard_pulpino is describing how to install the ZEPHYR_GCC_VARIANT=riscv32 toolchain.

The toolchain ZEPHYR_GCC_VARIANT=gccarmemb could be used by several boards, but there is no way for an ARM user to discover that he has more options than ZEPHYR_GCC_VARIANT=zephyr.

I don't have any examples of toolchain documentation being duplicated, but obviously the next board that supports espressif or riscv32 will have to duplicate the documentation as there is nowhere to put external toolchain documentation today.

pfalcon, 2017-09-18:
Thanks, that clarifies the issue (and hopefully will help to set a priority for it).

nashif, 2019-02-19:
over 1 year, closing.

closed 2019-02-19 by nashif

SebastianBoe, 2019-02-19:
@dbkinder : FYI