zephyrproject-rtos/zephyr #1538: esp32: ZEPHYR_GCC_VARIANT=espressif is broken for the latest esp-idf version

SebastianBoe, 2017-09-18:
Following the ESP32's instructions for installing and using the espressif SDK resulted in a broken build.

Specifically, I get the below error message when building hello_world.

Git blame shows that the offending line was added in mid-august, so I am guessing that the latest ESP-IDF SDK is not compatible with Zephyr. Haven't confirmed this by going back in time yet though ...

h3. Solution suggestions

1. Put a specific git revision in the SDK installation instructions.
2. Fix the latest ESP-IDF version to be compatible with Zephyr and then set up CI to ensure the latest ESP-IDF is always compatible with Zephyr.

`  /home/sebo/workspace_zephyr/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -Wp,-MD,drivers/gpio/.gpio_esp32.o.d  -nostdinc -isystem /home/sebo/workspace_zephyr/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/include -isystem /home/sebo/workspace_zephyr/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/include-fixed -I/home/sebo/zephyr/kernel/include -I/home/sebo/zephyr/arch/xtensa/include -I/home/sebo/zephyr/arch/xtensa/soc/esp32 -I/home/sebo/zephyr/boards/xtensa/esp32  -I/home/sebo/zephyr/include -I/home/sebo/zephyr/include -I/home/sebo/workspace_zephyr/zephyr/samples/hello_world/outdir/esp32/include/generated -include /home/sebo/workspace_zephyr/zephyr/samples/hello_world/outdir/esp32/include/generated/autoconf.h  -I/home/sebo/workspace_zephyr/esp/esp-idf/components/esp32/include -I/home/sebo/workspace_zephyr/esp/esp-idf/components/soc/esp32/include -I/home/sebo/zephyr/lib/libc/minimal/include  -I/home/sebo/zephyr/drivers/gpio -Idrivers/gpio -DKERNEL -D__ZEPHYR__=1 -c -g -std=c99 -Wall -Wformat -Wformat-security -D_FORTIFY_SOURCE=2 -Wno-format-zero-length -Wno-main -ffreestanding -Os -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-stack-protector -ffunction-sections -fdata-sections -fms-extensions -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Wno-pointer-sign -fno-strict-overflow -Werror=implicit-int   -I/home/sebo/zephyr/include/drivers   -I/home/sebo/zephyr/drivers    -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(gpio_esp32)"  -D"KBUILD_MODNAME=KBUILD_STR(gpio_esp32)" -c -o drivers/gpio/gpio_esp32.o /home/sebo/zephyr/drivers/gpio/gpio_esp32.c
In file included from /home/sebo/workspace_zephyr/esp/esp-idf/components/soc/esp32/include/soc/dport_access.h:20:0,
                 from /home/sebo/workspace_zephyr/esp/esp-idf/components/soc/esp32/include/soc/dport_reg.h:20,
                 from /home/sebo/zephyr/drivers/gpio/gpio_esp32.c:9:
/home/sebo/workspace_zephyr/esp/esp-idf/components/esp32/include/esp_dport_access.h:15:23: fatal error: sdkconfig.h: No such file or directory
`

SebastianBoe, 2017-09-18:
Can confirm that it is a regression from commit cb31222e in the esp-idf repo.

git checkout cb31222e~1

resolves the issue.

pfalcon, 2017-09-18:
Don't we pin esp-idf to a specific revision? Everybody does, it's still largely WIP and incompatible changes happen regularly (from what I heard).

SebastianBoe, 2017-09-18:
No.

The board's documentation says this:

> Two components are required in order to build this port: the `toolchain`_
> and the `SDK`_.  Both are provided by the SoC manufacturer.
> 
> The SDK contains headers and a hardware abstraction layer library
> (provided only as object files) that are required for the port to
> function.
> 
> The toolchain is available for Linux, Windows, and Mac hosts and
> instructions to obtain and set them up are available in the ESP-IDF
> repository, using the toolchain and SDK links above.
> 

Following the SDK link eventually gives you:

> git clone --recursive https://github.com/espressif/esp-idf.git

So latest master of esp-idf.

I would suggest pinning the version.

SebastianBoe, 2017-09-18:
Could you elaborate on it being WIP? Is there anything I should expect to be broken with esp32?

Specifically, is it expected that 

sanitycheck -p esp32 

will have 0 failures?

lpereira, 2017-09-18:
I believe that @pfalcon was saying that esp-idf is WIP, so likely to break every now and then.  I agree with the suggestion to pin to a particular revision, as we don't really care about esp-idf itself: we only need it for some headers and libhal.a, which are unlikely to significantly change.

pfalcon, 2017-09-18:
> I believe that @pfalcon was saying that esp-idf is WIP

Yes, that's what I meant.

andrewboie, 2017-09-19:
I would be not at all opposed to checking in a 'known good' version of ESP-IDF directly into the Zephyr tree, along with our other HALs in ext/ if there are no license issues.

SebastianBoe, 2017-09-19:
For source code this is OK IMHO, but not for the 452K binary libhal.a.

I worry it would bloat git over time.

lpereira, 2017-09-19:
Yeah.  The whole repository for esp-idf is licensed under an Apache License 2.0 (IIRC), so in theory we could do that.  But I'm not keen on distributing a binary blob with Zephyr.

pfalcon, 2017-09-19:
@nashif already raised (or mentioned) an issue that 90% (if not 95%) of Zephyr's "weight" are existing vendor libs in ext/ . So, if esp32 doesn't have that, it can be a fresh breath and start of something new (I for example would really ponder of using git submodules).

PR #4224 (2017-10-09): boards: esp32: Add note about fetching an older version of ESP-IDF
ESP-IDF is in constant development and it's likely that files Zephyr
depends on will be moved, removed, or renamed.  Make a note that an
older version of ESP-IDF should be used instead.

Closes #1538.
commit 357d6004 (2017-10-09): boards: esp32: Add note about fetching an older version of ESP-IDF

ESP-IDF is in constant development and it's likely that files Zephyr
depends on will be moved, removed, or renamed.  Make a note that an
older version of ESP-IDF should be used instead.

Closes #1538.
closed 2017-10-09 by nashif