zephyrproject-rtos/zephyr #1544: regression: net: K64F: DHCP seems to fail a lot after 91041f9e

mike-scott, 2017-09-19:
It's not 100% but most of the time, DHCP seems to fail on master branch using K64F.

I'm testing a revert on 91041f9e7363e5ccd984dfef7adbe9bd4b84b572 to see if that restores normal behavior.

pfalcon, 2017-09-19:
> It's not 100% but most of the time, DHCP seems to fail on master branch using K64F.

Given that DHCP [server] is an external thing, probably worth mentioning your setup. (Test process too.)

mike-scott, 2017-09-19:
Let me draw up a detailed report.  When I tested the DHCPV4 sample just now, I wasn't able to reproduce.  But our CI testing suddenly stopped working on K64F today after the mainline merges this morning.  And the logs showed that the devices weren't getting DHCP addresses anymore.

jukkar, 2017-09-20:
Ok, lets revert the multicast IPv4 mac address setting then. I will send a patch for that.

pfalcon, 2017-09-20:
@jukkar: I'd say we wait for additional info from @mike-scott. I eyeballed that patch and don't see anything obvious there (single bit flip is hard to blame without more arguments why would it matter). So far, @mike-scott reported that CI got broken, and well, that's normal, CIs are their to break (on their own) ;-).

jukkar, 2017-09-20:
@pfalcon I hope @mike-scott could test the #1564 and report if it is fixing things. And if everything works ok without the bit flip (the original way), then we can go back how the thing worked originally.

pfalcon, 2017-09-20:
Sure, sounds good. (My particular interest is due to the queued work on supporting true IPv6 multicast. Currently, the frdm_k64f eth is just in promisc mode overall; just imagine what kind of can of worms can be opened if that switched to the actual multicast filtering - unless we really try to figure out things).

mike-scott, 2017-09-21:
@jukkar Sorry, I forgot to write back earlier.  The new patch fixes CI testing as well.

commit 93695386 (2017-09-26): net: ethernet: Set dest MAC address for IPv4 mcast

If the ethernet packet destination MAC address is NULL when sending
IPv4 multicast or broadcast packet, then we must set it as otherwise
we might to access NULL pointer data.

Fixes #1544
closed 2017-09-26 by jukkar