zephyrproject-rtos/zephyr #1550: problem with pci_bus_scan resulting in an endless loop

eyuval, 2017-09-19:
I am using the standard paradigm for calling pci_bus_scan in a loop until it returns 0. However, in my scenario I  have several bars and I am hitting the condition 	'if (lookup.barofs >= max_bars)' at the end of pci_dev_scan, which resets the barofs and baridx for lookup.  What happens then, is that my pci_bus_scan loop simply restarts with bar 0 of the same device in an endless loop.

nashif, 2018-07-25:
@dcpleung since you are looking at PCI now, can you look into this and see if it is valid and if it can be fixed?

dcpleung, 2018-07-25:
For this behavior to manifest, the input parameter must have contained PCI_BAR_ANY, which will give you any BAR the pci_bus_scan() (and thus pci_dev_scan()) can find. Moreover, the current design does not guarantee a later call to pci_dev_scan() in the same thread will continue from previous call. For example, if another thread is scheduled and pci_dev_scan() is called, the global lookup struct will probably be changed.

So, in order to achieve the behavior to "scan until all BARs have exhausted", it requires re-working the flow to do external state tracking (e.g. having lookup struct per thread). The changes will not be trivial.

dcpleung, 2018-08-02:
Took a look at it more. The currently pci_dev_scan() does not necessarily give you BAR #0 the first time it is called on a device/func pair. If the function is previously used, the internal struct is changed. So, for example, calling this function again with PCI_ANY_BAR may give you BAR #3 instead of #0.

PR #14237 (2019-03-11): pci: properly scan all device resources
The change in commit e5349d74ab12d3a4c6c87f8723ed3c9a5355ae1b
results in only one resource being returned for a device with
pci_bus_scan(). The root cause for that issue was actually
because of wrapping around when scanning through BARs, where
BARs were scanned 0->1->2->..->6->0->1->.. in an endless loop
for a single bus:dev.function. So revert that commit and put
in a fix by moving on to the next function after going
through all the BARs.

Fixes #1550
commit c3bbfeb7 (2019-03-11): pci: properly scan all device resources

The change in commit e5349d74ab12d3a4c6c87f8723ed3c9a5355ae1b
results in only one resource being returned for a device with
pci_bus_scan(). The root cause for that issue was actually
because of wrapping around when scanning through BARs, where
BARs were scanned 0->1->2->..->6->0->1->.. in an endless loop
for a single bus:dev.function. So revert that commit and put
in a fix by moving on to the next function after going
through all the BARs.

Fixes #1550
closed 2019-03-11 by andrewboie