zephyrproject-rtos/zephyr #1558: Master reports itself as if it was 1.9.0 release

pfalcon, 2017-09-19:
Aka, we need PATCHLEVEL = 99.

nashif, 2017-09-19:
considering to call this 1.10.0-dev or something like that, 1.9.99 while seems not realistic, is a valid dot release that we might reach at some point :-), thoughts?

pfalcon, 2017-09-20:
Thoughts are: 1.9.99 is not realistic patch release we can reach. However, something like 5.13.99 in a dozen of years may be. If we want to solve that right now without much bikeshedding, let's just use 1.9.999. But that scheme worked pretty well for few releases, so I don't think it should be abandoned out of the blue. (And yep, the whole idea behind it was to be able to detect anything below a release X with just numeric comparisons (e.g. via C preprocessor).

pfalcon, 2017-09-20:
@carlescufi : ^^^, FYI. I believe it was you who raised concerns with "99", I apologize if it was another Nordic guy. Please see if above works for you.

carlescufi, 2017-09-20:
I am OK with using 999, the requirements on my side where to have a "build type" (dev, rc or production), which won't be solved by this anyway and it's tracked in an issue

pfalcon, 2017-09-20:
> which won't be solved by this anyway and it's tracked in an issue.

> In fact, with the current macros I actually prefer the 999 scheme, because at least one knows that this is a dev build without having to parse a string

Exactly, and at the corresponding ticket, I gave an idea that we can have:

~~~
#if PATCHLEVEL == 99(9)
#define BUILDTYPE "dev"
#endif
~~~

pfalcon, 2017-09-20:
Thanks for acking "999" btw.

PR #4138 (2017-10-02): release: Update version to 1.9.99 for pre-release of 1.10
Fixes #1558
commit b307ddff (2017-10-02): release: Update version to 1.9.99 for pre-release of 1.10

Fixes #1558
closed 2017-10-02 by nashif