zephyrproject-rtos/zephyr #1583: NFFS requires 1-byte unaligned accesses to flash

carlescufi, 2017-09-21:
Documented here: https://issues.apache.org/jira/browse/MYNEWT-13

It won't work on devices that cannot access the flash at any address and with 1-byte granularity

nashif, 2018-05-15:
lowering, a limitation rather than a bug

carlescufi, 2018-11-30:
CC @utzig

d3zd3z, 2019-12-03:
I fear this limitation is going to become more significant. I'm already seeing several devices that have a write alignment of 512. I don't know if this can be address with NFFS, or if there is possibly another filesystem that might handle these devices?

carlescufi, 2019-12-03:
> I fear this limitation is going to become more significant. I'm already seeing several devices that have a write alignment of 512. I don't know if this can be address with NFFS, or if there is possibly another filesystem that might handle these devices?

AFAIK NFFS hasn't seen any work recently and we should probably deprecated. Instead littlefs, which is already integrated into Zephyr, should be a good replacement. I assume that littlefs doesn't have this limitation @pabigot @nvlsianpu ?

pabigot, 2019-12-03:
littlefs allows you to specify the size (and implied alignment) of both read and program operations.  https://github.com/zephyrproject-rtos/zephyr/blob/master/tests/subsys/fs/littlefs/src/testfs_lfs.h and related files show how to configure the filesystem.  The default configuration uses 16-byte transfers.

nvlsianpu, 2019-12-04:
> AFAIK NFFS hasn't seen any work recently and we should probably deprecate it. 

That's right. NFFS should be deprecated - it has serious bugs by design.

nvlsianpu, 2020-01-27:
NFFS support was removed #21793.

closed 2020-01-27 by nvlsianpu