zephyrproject-rtos/zephyr #4066: Function typedef issue when generating htmldocs.

m-kru, 2017-09-24:
While generating htmldocs we can see some errors about function typedefs with following messages:
> Invalid definition: Expected identifier in nested name.

Here is more robust example. I have declared following typedef:

    typedef void (*wdt_callback_t)(struct device *dev, void *handle);

When I try to build htmldocs I get some errors with following structure:

>      Invalid definition: Expected identifier in nested name. [error at 4]
>      void(* wdt_callback_t) (struct device *dev, void *handle)
>      ----^
There are also some issues for http_send_data_t:

    int(* http_send_data_t) (struct net_pkt *pkt, net_context_send_cb_t cb, s32_t timeout, void *token,       void *user_data)

>      If declId, parameters, and qualifiers:
>       Invalid definition: Expecting "," or ")" in parameters_and_qualifiers, got "p". [error at 41]
>          int(* http_send_data_t) (struct net_pkt *pkt, net_context_send_cb_t cb, s32_t timeout, void  *token, void *user_data)
 >         -----------------------------------------^

dbkinder, 2017-09-26:
Alas, there's an open issue with the sphinx/breathe documentation tools (https://github.com/sphinx-doc/sphinx/issues/2683) where unnamed structures, in particular in nested unions / structs, generate this warning.  In our Zephyr documentation world if the unnamed structure can't be avoided, we add an entry to our filter list (.known-issues) to filter out "expected" warnings like this so our CI tools don't flag this as a problem.

dbkinder, 2017-09-26:
@m-kru Where are you seeing this warning show up? (Which PR?) With PR #1592,  ``make htmldocs`` should filter out "expected" warnings like this.)

m-kru, 2017-09-27:
I saw it on master.

tbursztyka, 2017-09-27:
It's a common issue, this willl require a fix in .known-issues/doc/networking.conf

dbkinder, 2018-02-09:
This is a reported issue with doxygen/breathe that we're unable to fix ourselves.  Workaround is to give unnamed nested structs and unions a name, or to add an exception to the warning message using the .known-issues mechanism we created to hide "expected" errors.

dbkinder, 2018-05-30:
Workaround provided until (if) Sphinx/Breathe addresses the unnamed nested structs/unions problem.

closed 2018-05-30 by dbkinder