zephyrproject-rtos/zephyr #4068: [BLE, nRF51822] Error -ENOMEM when use  bt_gatt_write_without_response function

iotvietmember, 2017-09-25:
Hi all,

My project uses Nordic nRF51822 chip (ARM) with Bluetooth Low Energy feature.

`nRF51822` is set to `Central` role in Connection model.

I tried to use `bt_gatt_write_without_responese` function to send data to a BLE peripheral device but this function return an error, code is -ENOMEM (-12).

Thanks for reading!

carlescufi, 2017-09-25:
@vudentz any thoughts?

Vudentz, 2017-09-26:
Could it be that the application wants to write more than it fits on a buffer:

	if (len + sizeof(op) > att->chan.tx.mtu) {
		BT_WARN("ATT MTU exceeded, max %u, wanted %zu",
			att->chan.tx.mtu, len + sizeof(op));
		return NULL;
	}

If that is really the problem we could fragment the data into multiple commands but the application can already check how big is the MTU with bt_gatt_get_mtu. Alternatively, we can make bt_gatt_write_without_response return how much it could write thus the user can decide to continue or not.

iotvietmember, 2017-09-26:
In `att.c` file:

Function 
```
struct net_buf *bt_att_create_pdu(struct bt_conn *conn, u8_t op, size_t len)
{
	struct bt_att_hdr *hdr;
	struct net_buf *buf;
	struct bt_att *att;

	att = att_chan_get(conn);
	if (!att) {
		return NULL;
	}

	if (len + sizeof(op) > att->chan.tx.mtu) {
		BT_WARN("ATT MTU exceeded, max %u, wanted %zu",
			att->chan.tx.mtu, len + sizeof(op));
		return NULL;
	}

	buf = bt_l2cap_create_pdu(NULL, 0);

	hdr = net_buf_add(buf, sizeof(*hdr));
	hdr->code = op;

	return buf;
}
```

Error at `buf = bt_l2cap_create_pdu(NULL, 0);` line

Follow the above function, in `conn.c`, error at `buf = net_buf_alloc(pool, K_FOREVER);` (line 1923)
-->
In `buf.c`, error at `buf = k_lifo_get(&pool->free, timeout);` (line 162)

Vudentz, 2017-09-26:
> Error at buf = bt_l2cap_create_pdu(NULL, 0); line
So is bt_gatt_write_without_response called from ISR context then? That would be a bug in the application since any API that do end up calling net_buf_alloc with K_FOREVER can fail. Though there are places where we don't actually check the return so we have some fixing to do as well.

iotvietmember, 2017-09-26:
@Vudentz Many thanks for your support. I have just fixed the bug in my application. I use a k_delay_work to execute the bt_gatt_write_without_response function from an ISR.

closed 2017-09-26 by iotvietmember