zephyrproject-rtos/zephyr #4131: gen_syscalls.py may choke on non-ascii chars

pfalcon, 2017-10-01:
We're having failures in our CI while building 3rd-party applications for Zephyr, e.g. https://ci.linaro.org/job/lite-aeolus-micropython/103/PLATFORM=qemu_x86,ZEPHYR_GCC_VARIANT=zephyr,label=docker-xenial-amd64-13/console :

~~~
00:03:48.643   CHK     include/generated/syscall_list.h
00:03:48.664 While parsing i2s.h
00:03:48.664 Traceback (most recent call last):
00:03:48.664   File "/home/buildslave/workspace/lite-aeolus-micropython/zephyr-rtos/scripts/gen_syscalls.py", line 244, in <module>
00:03:48.664     main()
00:03:48.664   File "/home/buildslave/workspace/lite-aeolus-micropython/zephyr-rtos/scripts/gen_syscalls.py", line 197, in main
00:03:48.664     syscalls = analyze_headers(args.include)
00:03:48.664   File "/home/buildslave/workspace/lite-aeolus-micropython/zephyr-rtos/scripts/gen_syscalls.py", line 100, in analyze_headers
00:03:48.664     for mo in api_regex.finditer(fp.read())]
00:03:48.664   File "/usr/lib/python3.5/encodings/ascii.py", line 26, in decode
00:03:48.664     return codecs.ascii_decode(input, self.errors)[0]
00:03:48.664 UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 988: ordinal not in range(128)
00:03:48.667 make[3]: *** [include/generated/syscall_list.h] Error 1
~~~

Now, I can't reproduce it locally at the 1st try, and our builds are broken for 3rd day, and the fact that nobody rushed to fix it, likely means that nobody else experiences it so far too.

But, i2s.h actually has non-ascii chars in comments, so there's something to do about this issue for sure.

dbkinder, 2017-10-01:
You might want to tweak the python scripts to be UTF-8 aware too.

pfalcon, 2017-10-01:
So, something really word happens here (there, on our CI systems). First of all, the issue indeed doesn't happen locally - I verified that i2s.h is being processed, and no error happens. It looks as if on the CI system, the file content gets decoded as if had "ascii" decoding. On my local system, it's:

~~~
$ python3 -c "import sys; print(sys.getdefaultencoding())"
utf-8
~~~

So, I ran the same command on the CI system and the output is the same. And yet file is being read with ascii encoding, as I can reproduce the issue locally if I execute:

~~~
>>> open("i2s.h", encoding="ascii").read()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.5/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 988: ordinal not in range(128)
~~~

pfalcon, 2017-10-01:
D'oh, turns out, Python's open() doesn't use sys.getdefaultencoding(), it uses locale.getpreferredencoding(False) as the default encoding: https://docs.python.org/3/library/functions.html#open . That apparently is inherited from libc's LOCALE, and can be funky things indeed. I'll fire up a patch to specify encoding explicitly in our scripts.

PR #4136 (2017-10-03): scripts: gen_syscalls: Use explicit encoding when reading headers
In Python, if open() doesn't specify "encoding" parameter,
locale.getpreferredencoding(False) will be used as the default,
as explained in
https://docs.python.org/3/library/functions.html#open ,
which may differ from system to system. So, explicitly specify
"encoding" param in open() call.

Also, fix a typo in a comment.

Fixes #4131.
closed 2017-10-03 by andrewboie