zephyrproject-rtos/zephyr #4150: USB device stack and class drivers

jfischer-phytec-iot, 2017-10-02:
# USB device stack and class drivers
This issue is there to provide an overview and to coordinate the work on USB device stack and the associated class drivers.

## Reusable USB components under samples
(_Components which can be moved to subsys/usb_)
- [ ] USB DFU class, inside PR  #716, developer: jfischer, status: ?
- [ ] webusb, ...
- [ ] wpanusb, ...

## Device stack
- [ ]  composite device support and common device descriptor, PR #716, developer: jfischer, status: WIP/review

## Class driver
- [ ] CDC and MSC Class Header, PR #4073, developer: jfischer, status: review
- [ ] CDC ACM, PR  #4147, developer: finikorg, status: WIP
- [ ] RNDIS, PR ?, developer: finikorg, status: planned?
- [ ] CDC EEM, PR?, #4144, developer: loicpoulain, status: planned?

### Device driver
(_OT but for the completeness_)
- device driver for Kinetis USBFSOTG Controller #542
- support for STM32Cube HAL_PCD USB driver #1384
- usb_dw: USB DMA #852

finikorg, 2017-10-03:
The PR #4147 is about CDC ECM

jfischer-phytec-iot, 2017-10-04:
@finikorg fixed

ydamigos, 2017-10-17:
PR #4376 adds DT support for STM32 USB device.

ydamigos, 2017-10-18:
I don't think we should cross out STM32Cube HAL_PCD USB driver. The driver, now, supports only F4 family. I will create a new PR to make driver generic based on #1384.

ydamigos, 2017-11-26:
PR #5163 adds support for the rest of the STM32 families.

sathishkuttan, 2018-07-17:
I'd like to see whether there is interest in developing an audio class (2.0 or even 3.0) stack. Please let me know.

carlescufi, 2018-10-17:
@sathishkuttan there is indeed. We talked about this today with @pawelzadrozniak and @finikorg and we'd have interest in sharing the work.

aurel32, 2018-10-31:
I am also interested in an USB audio class in Zephyr to run on a SAM E70 board. I would be happy to help or to test code.

larsgk, 2019-01-16:
Regarding WebUSB, I think it would be awesome if moved to subsys/usb with a few adjustments:

- [ ] Functionality to change manufacturer and product string
- [ ] Functionality to change serial runtime, e.g. to assign unique IDs from hardware
- [ ] The possibility to merge with other interfaces (e.g. CDC & WebUSB)
- [ ] License fix -> Apache2 (check with Intel?)

Also, it would be a good idea to involve Reilly Grant ( @reillyeon )

larsgk, 2019-01-16:
created this for WebUSB: #12527

carlescufi, 2019-03-05:
Closing this one since there is already an issue for WebUSB and USB is already up and running and fully functional.

closed 2019-03-05 by carlescufi