zephyrproject-rtos/zephyr #4151: Support gcov coverage reports

nashif, 2017-10-02:
We need to verify test coverage and increase it using existing tools designed for this. Gcov does provide this and relies on data generated by the compiler. Supporting this in an RTOS is not as easy as with a program running on the host system, so some work needs to be done on Zephyr to allow data collection from the target devices:

For reference, see:
- https://www.thanassis.space/coverage.html
- https://mcuoneclipse.com/2014/12/26/code-coverage-for-embedded-target-with-eclipse-gcc-and-gcov/
- https://github.com/reeteshranjan/libgcov-embedded

xasc, 2017-10-07:
Hey all,
Adding support for code coverage on the targets sounds very interesting. I would like to work on it.

Is there anybody I could discuss it further?

nashif, 2017-10-09:
@xasc if you want to discuss things related to this feature, this is the right place.

closed 2019-01-18 by nashif