zephyrproject-rtos/zephyr #4162: build error in http_get sample

andrewboie, 2017-10-03:
```
  CC      subsys/net/lib/sockets/sockets.o
/projects/zephyr2/subsys/net/lib/sockets/sockets.c: In function ‘zsock_sendto’:
/projects/zephyr2/subsys/net/lib/sockets/sockets.c:277:11: error: ‘err’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
   errno = -err;
           ^~~~
cc1: all warnings being treated as errors
make[7]: *** [/projects/zephyr2/scripts/Makefile.build:183: subsys/net/lib/sockets/sockets.o] Error 1
  CC      subsys/net/lib/sockets/getaddrinfo.o
make[7]: Target '__build' not remade because of errors.
make[6]: *** [/projects/zephyr2/scripts/Makefile.build:357: subsys/net/lib/sockets] Error 2
make[6]: Target '__build' not remade because of errors.
make[5]: *** [/projects/zephyr2/scripts/Makefile.build:357: subsys/net/lib] Error 2
```

This is when building for qemu_x86.
Not sure why CI hasn't caught this.

andrewboie, 2017-10-03:
@pfalcon feel free to reassign as appropriate

nashif, 2017-10-03:
introduced by 9ec60774f8cfda70fcdd3e39dc45df40eed3cb21

pfalcon, 2017-10-03:
I'm on it.

PR #4167 (2017-10-03): net: sockets: Simplify if logic in sendto(), fix warning
send()/sendto() aren't "front facing" functions, so when user calls
them, context type hopefully should be already validated by other
functions. They are also on critical path of app/network performance,
so getting rid of extra check helps a little bit too. This also
fixes a warning of "err" possibly being used non-initialized.

Fixes #4162
closed 2017-10-03 by nashif